MagicMirrorOrg/MagicMirror #860: Missing translation for 'Week' in module 'clock' when showWeek is set to true

BerndKohl, 2017-04-19:
**Platform**: Raspberry Pi 3
**Node Version**: 6.10.2
**MagicMirror Version**: 2.1.1
**Description**: Missing translation of "Week" in module 'clock' when showWeek is set to true.
**Steps to Reproduce**: Enable display of calendar week in module 'clock'
**Expected Results**: Display "Woche 16" when language is set to 'de' in config.js
**Actual Results**: "Week 16" is displayed

roramirez, 2017-04-19:
Send a Pull Request with this change for de language.

BerndKohl, 2017-04-20:
Would love to, simply don't know how to do this. Is there a short tutorial/introduction available?

BerndKohl, 2017-04-20:
As I can see this has already been fixed in development branch. Sorry for the hussle.

closed 2017-04-20 by BerndKohl