MagicMirrorOrg/MagicMirror #864: Timezone setting in Office 365 causes error in calendar module

Hermpi, 2017-04-21:
Hi, 

I have is some troubles to get my private calendar displaying on the MM and I think there is something wrong with Ical handler. For all additional information see my discussion with Cohron on the forum:
https://forum.magicmirror.builders/topic/2347/office-365-calendar-issues/8 

I'm able to get all my appointment showing up at the screen but the they all have a invalid value: "End in Invalid date". Only if I set my setting in Office 365 to timezone: UTC "Coordinated Universal Time" or set the appointment as All day event, the date is showing correctly. I looked in the ical files and the only difference I found between the working and not working appointments are the DTSTART & DTEND. I know this the line were the actually timing data is come from and the following line seems not to work:
DTSTART;TZID="(UTC+01:00) Amsterdam, Berlijn, Bern, Rome, Stockholm, Wenen":2
 0170418T080000 
but this will work [time zone set to UTC]: 
DTSTART;TZID=UTC:20170422T083000 

Could it be that the code that gets the data from DTSTART isn't able subtract the timing because the timingzone is between brackets? 

Running MM v2.1.0 on Raspi 3

tgabben, 2017-05-07:
I'm seeing the same symptom. I tried to feed the calendar module an O365 .ics file, and it's showing me a string of appointments in the past as 'upcoming' and all with the 'Ends in Invalid Date'.

It does appear that all of these unwanted (and long passed) events were created in a time zone other than my own (I've even noticed that there is one 'legitimate' event which is actually in the future, mixed into the rest of the list and still indicated as 'Ends In Invalid Date'.) The only appointment which appears to be displaying correctly is one that I have upcoming, in town (as such, was created in my time zone).

Hermpi, 2017-05-08:
They have already did a Pull request to fix this issue.  My O365 calendar works perfect now. 

See: https://github.com/MichMich/MagicMirror/pull/875 

I think they will implement it with the next update for MM. I replace the two files (ical.js & node-ical.js) manually.

tgabben, 2017-05-08:
Unfortunately, those modified ical.js & node-ical.js files don't address the issue I'm having with events persisting in the display with "Ends in Invalid date". The original issue does appear to have been related to Exchange calendars, but I'm not seeing any relief from what appears to be an issue with the TZID format.

Hermpi, 2017-05-09:
Okay I get what your problem is. I delete all events in the past from my calendar to be sure this doesn't cause a problem. I my case there were upcoming and past events displayed with the invalid date error. I will add a event in the past in my calendar and when I come home from work I will check what the MM displayed.

tgabben, 2017-05-09:
Ah - unfortunately, clearing all past events isn't entirely practical or feasible for me.

Additionally - as you've acknowledged - the problem exists for upcoming events as well as past events. Not only are there past meetings junking up the list, but there are one or two upcoming meeting invites in there, too - showing as "Ends in Invalid date" because they were created in Eastern Time by a colleague in that time zone.

B4ttleduck, 2017-08-03:
Same problem here, not O365 but the regular outlook Calendar.
Replacing the 2 files didn't work :(

loctruong96, 2017-09-09:
I am having the exact same problem as you guys are having and replacing the new files update didn't help either. However, I have found a reasonable workaround for this problem.
1. Import office 365 calendar into a google calendar, this will take care of past events
2. go to https://flow.microsoft.com/ and setup a flow trigger that will create a new event in google calendar whenever an event in office 365 is created. guide example (https://shift.newco.co/sync-your-calendars-using-microsoft-flow-and-yes-google-calendar-works-too-a28be5a604dd)
3. Go to google calendar and create a private address to use in the calendar module for magic mirror.
NOTE: Trigger will run every 5 minutes. Make sure to give it sometimes after you create a new test event in office 365 to appear in google calendar. Past google calendar will NOT be created using flow, only future events.

MichMich, 2017-09-29:
Great workaround by @loctruong96. We'll add it to the Wiki. If somebody wants to create a real fix, feel free to send a PR. Closing this issue for now.

closed 2017-09-29 by MichMich

pkkid, 2018-01-01:
To continue this discussion a bit. -- Will the workaround mentioned in @loctruong96's comment work if an even is updated (moved) to a new time?

Hermpi, 2018-01-03:
Hey guys, 

A couple weeks ago my Raspberry crashed and I had to reinstall everything included MM. The replacing of the two files didn't work anymore. So I did the same as @loctruong96 mentioned. This works great with one major downside! In my phone every events comes up twice.... 

I just switch jobs so I had to implement my new work calendar (regular outlook agenda). I will look at this in the weekend maybe I find some other workaround. I keep you updated ;)

tgabben, 2018-01-04:
@pkkid - the O365 workflow mentioned serves to create new events on a google calendar only - it has no knowledge of updates/changes, and doesn't alter existing events on the associated google calendar. Any modifications to events have to be made via the google calendar itself.

It's an imperfect workaround, but it seems to be all we've got for the moment.

pkkid, 2018-01-04:
FWIW, I made another workaround using the html link o365 provides. Perhaps I can make it a Herkou app or something if there is demand.

Essentially, I used an ajax call from the HTML o365 calendar that returns a list of all events in a given time frame in json format and recurring events denormalized. I wrote a quick passthough on my webserver that reads these events and serves a new ics calendar with normal formatting and dates. Magic Mirror seems to be reading them fine, and since it gets the most up to date information on o365 every time, it'll have all updates (not just creates). So it seems perfect as a read only setup.

If you're interested to try it out, send me a pm.