MagicMirrorOrg/MagicMirror #866: magicmirror 2

josgab, 2017-04-21:
0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@2.14.7
3 info using node@v4.2.1
4 verbose config Skipping project config: /home/pi/.npmrc. (matches userconfig)
5 verbose run-script [ 'prestart', 'start', 'poststart' ]
6 info prestart magicmirror@2.1.1
7 info start magicmirror@2.1.1
8 verbomose unsafe-perm in lifecycle true
9 info magicmirror@2.1.1 Failed to exec start script
10 verbose stack Error: magicmirror@2.1.1 start: `sh run-start.sh`
10 verbose stack Exit status 1
10 verbomose stack     at EventEmitter.<anonymous> (/usr/local/lib/node_modules/npm/lib/utils/lifecycle.js:214:16)
10 verbose stack     at emitTwo (events.js:87:13)
10 verbose stack     at EventEmitter.emit (events.js:172:7)
10 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/lib/utils/spawn.js:24:14)
10 verbose stack     at emitTwo (events.js:87:13)
10 verbose stack     at ChildProcess.emit (events.js:172:7)
10 verbose stack     at maybeClose (internal/child_process.js:818:16)
10 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
11 verbose pkgid magicmirror@2.1.1
12 verbose cwd /home/pi
13 error Linux 4.4.50-v7+
14 error argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
15 error node v4.2.1
16 error npm  v2.14.7
17 error code ELIFECYCLE
18 error magicmirror@2.1.1 start: `sh run-start.sh`
18 error Exit status 1
19 error Failed at the magicmirror@2.1.1 start script 'sh run-start.sh'.
19 error This is most likely a problem with the magicmirror package,
19 error not with npm itself.
19 error Tell the author that this fails on your system:
19 error     sh run-start.sh
19 error You can get their info via:
19 error     npm owner ls magicmirror
19 error There is likely additional logging output above.
20 verbose exit [ 1, true ]

fewieden, 2017-04-22:
which os are you using?

MichMich, 2017-04-22:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2017-04-22 by MichMich

Appleudaya, 2019-01-07:
the above comment is refered and run the commant . But, the MagicMirror is not auto starting automatically when the raspberry pi ON
So, please give the reference to Auto start tyhe MagicMirror