MagicMirrorOrg/MagicMirror #868: ETA of calendar module is not displayed correctly

JoDa92, 2017-04-22:
Some events (more than a month in the future) doesnt get displayed correctly..
Its a google calendar!

**Picture:**
http://i.imgur.com/5xyeN2p.jpg

**Data:**

BEGIN:VEVENT
DTSTART:20170627T080000Z
DTEND:20170627T090000Z
DTSTAMP:20170422T145944Z
UID: .....
CREATED:20170410T094531Z
DESCRIPTION:
LAST-MODIFIED:20170417T185228Z
LOCATION:
SEQUENCE:1
STATUS:CONFIRMED
SUMMARY: Test
TRANSP:OPAQUE
END:VEVENTNT

JoDa92, 2017-05-03:
The data is correctly parsed:

pi@raspberrypi:~/MagicMirror/modules/default/calendar$ node debug.js 
Create fetcher ...
Create fetcher done! 
[ { title: 'Test1',
    startDate: '1498550400000',
    endDate: '1498554000000',
    fullDayEvent: false,
    class: undefined,
    location: false,
    geo: false,
    description: false },
  { title: 'Test2',
    startDate: '1501495200000',
    endDate: '1501498800000',
    fullDayEvent: false,
    class: undefined,
    location: false,
    geo: false,
    description: false },
  { title: 'Test3',
    startDate: '1504821600000',
    endDate: '1504908000000',
    fullDayEvent: true,
    class: undefined,
    firstYear: 2014,
    location: false,
    geo: false,
    description: false },
  { title: 'Test4',
    startDate: '1505210400000',
    endDate: '1505214000000',
    fullDayEvent: false,
    class: undefined,
    firstYear: 2016,
    location: false,
    geo: false,
    description: false },
  { title: 'Test5',
    startDate: '1505728800000',
    endDate: '1505732400000',
    fullDayEvent: false,
    class: undefined,
    firstYear: 2017,
    location: false,
    geo: false,
    description: false },
  { title: 'Test6',
    startDate: '1507413600000',
    endDate: '1507500000000',
    fullDayEvent: true,
    class: undefined,
    firstYear: 2015,
    location: false,
    geo: false,
    description: false },
  { title: 'Test7',
    startDate: '1513119600000',
    endDate: '1513206000000',
    fullDayEvent: true,
    class: undefined,
    firstYear: 2014,
    location: false,
    geo: false,
    description: false } ]
------------------------------------------------------------

JoDa92, 2017-05-03:
If i try to use the console.log function in the calendar.js file it doesnt show me any info on the screen. Im starting it with npm start dev.

JoDa92, 2017-05-03:
The line
moment(event.startDate, "x").fromNow()
is making the startDate 1504821600000 to "in %s Mon."

JoDa92, 2017-05-04:
Check this out:
http://jsfiddle.net/rLjQx/1094/
without the "x" its working!!

Its also working if i use a "X" for the normal unix timeformat.

Im gonna test it later on my pi

HawkingJan, 2017-05-11:
I've the same problem. I think it might be an issue of an old moments.js version.
In your "working" example you use
http://cdnjs.cloudflare.com/ajax/libs/moment.js/2.2.1/moment.min.js
as external resource.

If I use
http://momentjs.com/downloads/moment.min.js
or
http://momentjs.com/downloads/moment.js
(which is currently equivalent to version 2.18.1)
instead
also

```
var now = moment(1504821600000 , "x").fromNow();
alert(now);
```
results in "in 4 months" as it should be.

My only problem is that in my installation the file in 
~/MagicMirror/node_modules/moment/moment.js
~/MagicMirror/vendor/node_modules/moment/moment.js
~/MagicMirror/node_modules/moment/min/moment.min.js
~/MagicMirror/vendor/node_modules/moment/min/moment.min.js
are already version 2.18.1 (at least that is written in their first lines). Therefore I don't know how I could update...
Can you confirm that moment version 2.18.1 solves this issue for your test project and do you know how to update the moment.js that is used by MagicMirror?

MichMich, 2017-05-15:
IS this still an issue? Or can I close this?

HawkingJan, 2017-05-15:
Yes, it is still an issue (for me). However it might depend on the moments version. I used a puplic calender that has only few entries so some are a few month in the future and erroneously shown as "in %s Mon.".

```

                {
                        module: "calendar",
                        header: "Feiertage",
                        position: "top_left",
                        config: {
                                calendars: [
                                        {
                                                url: "http://www.ifeiertage.de/by-.ics"
                                        }
                                ]
                        }
                },

```

Maybe this example helps you to reproduce this issue.

JoDa92, 2017-05-21:
Still an issue. I switched to absolute dates for the moment. Will have a look at it again in a few days!

MichMich, 2017-05-26:
Did this approach solve the issue for you? (In other words: can I close this issue?)

HawkingJan, 2017-05-28:
I wouln't call the switch to absolute dates a real solution for an issue that occurs on relative dates. So for me this issue is not solved.

HawkingJan, 2017-07-15:
I installed some general updates and new pageckages (
```
sudo apt-get install python-pip 
sudo pip install Flask
....
```
)
and now the issue is solved for me. Unfortunately I don't know which change exactly fixed this issue.. Maybe ghost can also update his system to check if also for him the issue is solved.

MichMich, 2017-09-29:
Closed due to inactivity.

closed 2017-09-29 by MichMich