MagicMirrorOrg/MagicMirror #876: Improve translation flexibility

Arkkimaagi, 2017-04-25:
String joining can get us only so far regarding translations. In some cases it would be easier to reorder words with the translation strings.

I'd suggest that we'd add support for something like string literals that are coming in future version of js.

For example calendar could support Finnish way of saying the `RUNNING` a bit better
en.js:
```js
	"RUNNING": "Ends in {timeUntilEnd}",
```
fi.js:
```js
	"RUNNING": "Päättyy {timeUntilEnd} päästä",
```
Notice the `{timeUntilEnd}` variable there? That will be replaced with the variable during translation.

This would be rendered in Finnish as:
```
Päättyy viiden minuutin päästä
```
Where moment.js returns `viiden minuutin` as we have five minutes left.

Here is a screenshot from my mirror (where I have dropped the capitalization with css):
[image]

Of course we'd want to keep the translator backwards compatible, we'd want also this to work just as it does now so that all the current implementations do not break:
```js
	"RUNNING": "Ends in",
```
It should fallback to current approach resulting 
```
Ends in five minutes
```

I have a working sample of this thing with an example for this calendar thing only, but support could be built for all translations in future to combat rigid string joining problems.

Arkkimaagi, 2017-05-01:
Seems that this has been merged into develop, maybe this issue can be closed? Or do you close these only when merged into master?

roramirez, 2017-05-05:
Hi @Arkkimaagi you should be close this issue.

Arkkimaagi, 2017-05-05:
Closing the issue as it was merged.

closed 2017-05-05 by Arkkimaagi