MagicMirrorOrg/MagicMirror #880: My Mirror broke at startup with pm2

ar84, 2017-04-29:
Hey Guys.

I had a Problem starting my MM.

Anyone knows what To do??

/home/pi/.pm2/logs/mm-error-0.log last 15 lines:
0|mm       |     at loadModules (/home/pi/MagicMirror/js/app.js:172:3)
0|mm       |     at /home/pi/MagicMirror/js/app.js:218:4
0|mm       |     at loadConfig (/home/pi/MagicMirror/js/app.js:77:4)
0|mm       |     at App.start (/home/pi/MagicMirror/js/app.js:206:3)
0|mm       | TypeError: Cannot read property 'split' of undefined
0|mm       |     at loadModule (/home/pi/MagicMirror/js/app.js:108:24)
0|mm       |     at loadNextModule (/home/pi/MagicMirror/js/app.js:161:5)
0|mm       |     at /home/pi/MagicMirror/js/app.js:163:6
0|mm       |     at Class.loaded (/home/pi/MagicMirror/modules/node_modules/node_helper/index.js:37:3)
0|mm       |     at loadModule (/home/pi/MagicMirror/js/app.js:144:6)
0|mm       |     at loadNextModule (/home/pi/MagicMirror/js/app.js:161:5)
0|mm       |     at loadModules (/home/pi/MagicMirror/js/app.js:172:3)
0|mm       |     at /home/pi/MagicMirror/js/app.js:218:4
0|mm       |     at loadConfig (/home/pi/MagicMirror/js/app.js:77:4)
0|mm       |     at App.start (/home/pi/MagicMirror/js/app.js:206:3)

0|mm       | Starting MagicMirror: v2.1.1
0|mm       | Loading config ...
0|mm       | Loading module helpers ...
0|mm       | Initializing new module helper ...
0|mm       | Module helper loaded: MMM-Admin-Interface
0|mm       | WARNING! Could not load config file. Starting with default configuration. Error found: TypeError: Cannot read property 'split' of undefined
0|mm       | Loading module helpers ...
0|mm       | Initializing new module helper ...
0|mm       | Module helper loaded: MMM-Admin-Interface
0|mm       | App threw an error during load
0|mm       | TypeError: Cannot read property 'split' of undefined
0|mm       |     at loadModule (/home/pi/MagicMirror/js/app.js:108:24)
0|mm       |     at loadNextModule (/home/pi/MagicMirror/js/app.js:161:5)
0|mm       |     at /home/pi/MagicMirror/js/app.js:163:6
0|mm       |     at Class.loaded (/home/pi/MagicMirror/modules/node_modules/node_helper/index.js:37:3)
0|mm       |     at loadModule (/home/pi/MagicMirror/js/app.js:144:6)
0|mm       |     at loadNextModule (/home/pi/MagicMirror/js/app.js:161:5)
0|mm       |     at loadModules (/home/pi/MagicMirror/js/app.js:172:3)
0|mm       |     at /home/pi/MagicMirror/js/app.js:218:4
0|mm       |     at loadConfig (/home/pi/MagicMirror/js/app.js:77:4)
0|mm       |     at App.start (/home/pi/MagicMirror/js/app.js:206:3)
0|mm       | Whoops! There was an uncaught exception...
0|mm       | TypeError: Cannot read property 'split' of undefined
0|mm       |     at loadModule (/home/pi/MagicMirror/js/app.js:108:24)
0|mm       |     at loadNextModule (/home/pi/MagicMirror/js/app.js:161:5)
0|mm       |     at /home/pi/MagicMirror/js/app.js:163:6
0|mm       |     at Class.loaded (/home/pi/MagicMirror/modules/node_modules/node_helper/index.js:37:3)
0|mm       |     at loadModule (/home/pi/MagicMirror/js/app.js:144:6)
0|mm       |     at loadNextModule (/home/pi/MagicMirror/js/app.js:161:5)
0|mm       |     at loadModules (/home/pi/MagicMirror/js/app.js:172:3)
0|mm       |     at /home/pi/MagicMirror/js/app.js:218:4
0|mm       |     at loadConfig (/home/pi/MagicMirror/js/app.js:77:4)
0|mm       |     at App.start (/home/pi/MagicMirror/js/app.js:206:3)

fewieden, 2017-04-29:
Please refer to https://forum.magicmirror.builders for support and troubleshooting.

ar84, 2017-05-03:
can be closed

closed 2017-05-03 by ar84