MagicMirrorOrg/MagicMirror #884: bash syntax problem with installer

z0diak, 2017-05-06:
Installer in https://github.com/MichMich/MagicMirror/blob/master/installers/raspberry.sh doesn't work correctly on a brand-new RPI with raspbian-jessie. 
Syntax error 
```
bash: line 153: syntax error near unexpected token `then'
bash: line 153: `then'
```

Going to propose PR to fix that.

Just thinking, maybe we should have tests running for installer? Just like: exit code is 0

roramirez, 2017-05-08:
There are a workaround where you can using the script installer like this.

`bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh)"`

If you want adding test for installer script ...go ahead.

MichMich, 2017-05-15:
Fix merged.

closed 2017-05-15 by MichMich

aramaki87, 2017-05-25:
Please also fix at https://magicmirror.builders

z0diak, 2017-05-26:
@aramaki87 what do you mean?

develuxe, 2017-06-05:
@z0diak He probably meant the tutorial here -> https://forum.magicmirror.builders/topic/236/complete-setup-tutorial/6

z0diak, 2017-06-05:
@develuxe Yes, I also thought this way, but since we've changed the code in way to keep execution command the same (not what was suggested by roramirez) there is no need to update tutorial

develuxe, 2017-06-05:
@z0diak I've found this thread because I had the exact same problem. I followed the linked tutorial. With the posted workaroung I dont get this error. Just fyi

joeframbach, 2017-06-28:
This is still an issue on a new Jessie installation using the install command on https://magicmirror.builders/

roramirez, 2017-07-06:
Hey Guys!, few minutes ago the install command onto the [website](https://magicmirror.builders/) are updated. Please you follow this sentence  👌