MagicMirrorOrg/MagicMirror #886: stuck in 'node:install.js'

puluter, 2017-05-07:
`Installing dependencies ...
npm WARN engine request@2.81.0: wanted: {"node":">= 4"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine feedme@1.0.0: wanted: {"node":">=0.12"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine grunt-jsonlint@1.1.0: wanted: {"node":">=4.0"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine http-auth@3.1.3: wanted: {"node":">=4.6.1"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine spectron@3.6.2: wanted: {"node":">=0.12.4"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine grunt-eslint@19.0.0: wanted: {"node":">=4"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine eslint@3.19.0: wanted: {"node":">=4"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine stylelint@7.10.1: wanted: {"node":">=4.2.1"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine apache-crypt@1.2.1: wanted: {"node":">=4.6.1"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine apache-md5@1.1.2: wanted: {"node":">=4.6.1"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine har-validator@4.2.1: wanted: {"node":">=4"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine hawk@3.1.3: wanted: {"node":">=0.10.32"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine form-data@2.1.4: wanted: {"node":">= 0.12"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine deep-extend@0.4.1: wanted: {"node":">=0.12.0","iojs":">=1.0.0"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine esprima@3.1.3: wanted: {"node":">=4"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine har-schema@1.0.5: wanted: {"node":">=4"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine hoek@2.16.3: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine boom@2.10.1: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine cryptiles@2.0.5: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine webdriverio@4.8.0: wanted: {"node":">= 0.12.0"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine deep-extend@0.4.1: wanted: {"node":">=0.12.0","iojs":">=1.0.0"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine co@4.6.0: wanted: {"iojs":">= 1.0.0","node":">= 0.12.0"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine get-stdin@5.0.1: wanted: {"node":">=0.12.0"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine string-width@2.0.0: wanted: {"node":">=4"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine cosmiconfig@2.1.3: wanted: {"node":">=0.12"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine postcss@5.2.17: wanted: {"node":">=0.12"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine is-fullwidth-code-point@2.0.0: wanted: {"node":">=4"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine strip-bom@3.0.0: wanted: {"node":">=4"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine shelljs@0.7.7: wanted: {"node":">=0.11.0","iojs":"*"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine esprima@3.1.3: wanted: {"node":">=4"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine esprima@3.1.3: wanted: {"node":">=4"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine string-width@2.0.0: wanted: {"node":">=4"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine is-fullwidth-code-point@2.0.0: wanted: {"node":">=4"} (current: {"node":"0.10.29","npm":"1.4.21"})
npm WARN engine co@4.6.0: wanted: {"iojs":">= 1.0.0","node":">= 0.12.0"} (current: {"node":"0.10.29","npm":"1.4.21"})
 
> electron@1.6.6 postinstall /home/pi/MagicMirror/node_modules/electron
> node install.js
`

I've been stuck in this page for such a long time,is there a solution?
Huge tks if there is one.

fewieden, 2017-05-07:
your node version is 3 years old update it and it will work fine

MichMich, 2017-05-15:
As fewieden says: please update node. Should be solved.

closed 2017-05-15 by MichMich