MagicMirrorOrg/MagicMirror #887: Automatic Installer & npm start (No widgets, not finding resources)

bogdan1133, 2017-05-07:
**Platform**: Raspberry Pi 3
**MagicMirror Version**: (v2).

**Description**: When I try automatic installer it and start with electron-chromedriver.js, it shows that electron-chromedriver.js registry 404. When manually downloading and performing npm install no widgets are displayed. After npm start dev (Development mode) it shows multiple errors that the modules are not found, even though its all there in the module default folder.

fewieden, 2017-05-07:
in the latest version there is a npm install task in the vendor folder as well which gets executed automatically after a successful install, yours wasn't successful so this was never executed

bogdan1133, 2017-05-08:
@fewieden So how can this be solved ? By performing a manual npm install in the vendor folder ? Cause I have tried multiple times to make it work, with manual and automatic installation.

MichMich, 2017-05-15:
Did you try the develop branch, or the master branch?

MichMich, 2017-05-26:
Closing due to inactivity.

closed 2017-05-26 by MichMich