MagicMirrorOrg/MagicMirror #889: Hidden module isn't completely hidden

tshort, 2017-05-14:
I have an MM setup with modules that overlap using the MMM-ProfileSwitcher and MMM-TouchNavigation modules. Here's a [forum link](https://forum.magicmirror.builders/topic/2566/fridge-mounted-display) on my touch-based, not-a-mirror setup.

I have two or more IFrames in the same area. When I hide the top one, the underlying module hides, but mouse clicks and touch don't work on any of the links shown in the visible IFrame. If I click on a position of a link on the top (invisible) module, it takes me to that link. This happens in Firefox and Chrome (I'm running in server mode).

MM currently shows and hides modules by toggling opacity:

https://github.com/MichMich/MagicMirror/blob/master/js/main.js#L193

that has:

```js
			moduleWrapper.style.opacity = 0;
```

That hides it visibly, but I don't think it hides the click areas.

I fixed this my local version by changing to the following in hideModule:

```js
			moduleWrapper.style.opacity = 0;
			moduleWrapper.style.position = "absolute";
			moduleWrapper.style.left = "-9000px";
```
I toggle position between "absolute" and "static" and toggle left between "-9000px" and "0px". This probably isn't the right way to do this generally (I don't understand the HTML DOM model very well), but it works for my uses.

MichMich, 2017-05-15:
I understand the issue and your approach probably saves it, but is indeed not the best way to do it. This needs a bit of a cleaner solution. Unfortunately, "touch" functionality is not a supported part of the core, so I don't expect anyone to solve this soon. Of course, you can send a PR with a clean solution. If it's well thought thru, I'll merge it in the develop branch.

Let me know if this answers your question so I can close this issue.

tshort, 2017-05-15:
That answers it. Thanks.

closed 2017-05-15 by MichMich