MagicMirrorOrg/MagicMirror #891: [Suggestion] Move custom.css to /config/

JanLoebel, 2017-05-17:
Hi,

what do you think about moving the "_/css/custom.css_" file to "_/config/custom.css_" and only load it if it exists? This would be great to have the possibility to use Docker and only specify the local config- and module-folder.

Any suggestions?

Best regards,
Jan

roramirez, 2017-05-18:
There not possible to mount css directory into volume of docker?

JanLoebel, 2017-05-18:
Not really, because in the css directory there is also the _main.css_ which you would have to copy in the mounted local directory.

MichMich, 2017-05-18:
What could be an option, is to make the main.css path configurable. With the default set to the current location. This way, you can modify it in your config, and it won't be a breaking change.

Feel free to send a PR for this. Make sure to update the CHANGELOG & README accordingly.

closed 2017-06-06 by JanLoebel