MagicMirrorOrg/MagicMirror #894: sendSocketNotification is not working from node_helper to main module

priya99, 2017-05-19:
Hi,

I am new to node.js.I am trying to create new sample module "test".I am sending notification from test module to node_helper and trying to receive notification back to test module.From test module to node_helper notification works fine, but test module does not receive any notification from node_helper.

Any help appreciated.
Here is my module code:
test.js:

Module.register("test",{

defaults:{
	text:"Happy Birthday!!!!"
},

getDom: function() {
	var wrapper = document.createElement("div");
	wrapper.innerHTML =this.config.text;
	return wrapper;
},

start:function(){
	this.sendSocketNotification("main",{message:"test1"});
},

socketNotificationReceived: function(notification, payload) {
	console.log('socketNotificationReceived() in main module..'+notification);
	if (notification === "test2") {
		console.log('test2 socket notification received...');
	}
}

});

Here is node_helper.js:

var NodeHelper = require("node_helper");

module.exports = NodeHelper.create({
// Override start method.
start: function() {

},

// Override socketNotificationReceived method.
socketNotificationReceived: function(notification, payload) {
	var self=this;
	console.log('*Starting socketNotificationReceived()*'+notification);
	if (notification === "main") {
		console.log(notification+' socket notification received...');
		this.function1();
	}
},

function1:function(){
	console.log('inside function1');
	this.sendSocketNotification("test2",{message:"test2"});
}

});

Here is config.js:

var config = {
port: 8080,
ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], // Set [] to allow all IP addresses.

language: "en",
timeFormat: 24,
units: "metric",

modules: [
	
	{
		module: "test",
		position: "top_right"
	}
    ]

};

MichMich, 2017-05-19:
Make sure you first send one message from the module to the helper. After that the communication channel is created and you can send a message from the helper to the module.

priya99, 2017-05-19:
Thanks for quick reply.I have created communication channel by sending notification from test module to node_helper.

`start:function(){
	this.sendSocketNotification("main",{message:"test1"});
},`

priya99, 2017-05-22:
I am stuck there since last 3-4 days.If something is missing or wrong please correct me.Any help appreciated.

fewieden, 2017-05-22:
There is no issue, you just did not understand that the module doesn't log in the same place as the node helper.

Node helper -> server -> logs into terminal
Module -> client -> logs into browser/electron

[image]

priya99, 2017-05-23:
Thanks  for the reply.

MichMich, 2017-05-26:
I guess it's solved :) Closing it for now ...

closed 2017-05-26 by MichMich