MagicMirrorOrg/MagicMirror #901: Model B+ supported?

dpapathanasiou, 2017-05-22:
I've seen issues and troubleshooting posts referring to getting MM running on a Model B+, so I'm not sure if it is officially supported.

The automated installer fails with this message:

<pre>Sorry, your Raspberry Pi is not supported.
Please run MagicMirror on a Raspberry Pi 2 or 3.
If this is a Pi Zero, you are in the same boat as the original Raspberry Pi. 
You must run in server only mode.</pre>

According to <tt>cat /proc/cpuinfo</tt>, my Pi is a [Model B+](http://www.raspberrypi-spy.co.uk/2012/09/checking-your-raspberry-pi-board-version/):

<pre>Hardware        : BCM2835
Revision        : 0010</pre>

But I'm not sure if that places it in the Pi 1 or 2/3 spectrum.

I'm running the full Raspbian Jessie (Pixel) image from 2017-04-10, so I do not know if I need to downgrade to Jessie Lite, and add the X11 components later.

nhubbard, 2017-05-23:
@dpapathanasiou The difference is mostly hardware changes, like more USB ports and better board arrangement. Unfortunately, that does place you within the same spectrum as the original Raspberry Pi.

roramirez, 2017-05-25:
@dpapathanasiou What output show you the command?

`uname -a`

The issue is  if the electron is available for the architecture of your device. If not only the only way is run [as _serveronly_ option](https://github.com/MichMich/MagicMirror#server-only)

You can follow and try the instructions in the installer script. If it work for you  send a PullRequest to fix this.

AFDM74, 2017-05-26:
@roramirez, Can you please explain more about the server mode?

I'm not sure if I got this right, correct me if I'm wrong.

As I'm understanding to make MagicMirror run in the RaspberryPi 1 (My model is the same as @dpapathanasiou) I will have to run it in server mode, because the Electron App is not available to this RPi model architecture. @MichMich writes that installations on the RPi 1 must be run in a Full Screen browser. And here is where I get a little confused (mostly because I don't know Electron). @MichMich reference to the Full Screen Browser is the Chromium or any other browser available in the LXDE Desktop Enviroment (or Pixel), right? And I will open the localhost server of the Magic Mirror server on the browser, correct? All I have to do is then autostart the browser in fullscreen to have the MagicMirror magic.

My "uname -a"  Linux magicmirror 4.9.24+ #993 Wed Apr 26 17:56:54 BST 2017 armv6l GNU/Linux

Thanks,
Deco Martins

MichMich, 2017-05-26:
The guide you are referring to is talking about version 1. (As stated above the guide). This is version 2 which focusses on RPi 3 but might work on the Pi2 B+. If you need help getting it up and running (for example using the server only mode), please check out http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2017-05-26 by MichMich

Force-United, 2019-01-25:
Hello!

I have same problem. But Raspberry Pi B+ is a Raspberry Pi 1 model, so that not working. you must use Raspberry Pi 3 Model B+ and Raspberry Pi 3 Model B working great.

Mvh
Force-United

Doublefire-Chen, 2022-11-23:
> so that not working. you must use Raspberry Pi 3 Model B+ and Raspberry Pi 3 Model B working great.

I also have a B+. Now the board is getting more and more expensive. I really hope this old stuff can be good used in Magic Mirror.