MagicMirrorOrg/MagicMirror #906: MMM-voice sendSocketNotification is not a function

ezar, 2017-05-28:
1|mm       | MMM-voice: Checking files.
1|mm       | MMM-voice: Starting pocketsphinx.
1|mm       | Whoops! There was an uncaught exception...
1|mm       | TypeError: this.sendSocketNotification is not a function
1|mm       |     at fs.appendFile (/home/pi/MagicMirror/modules/MMM-voice/node_helper.js:311:22)
1|mm       |     at /home/pi/MagicMirror/modules/MMM-voice/node_modules/graceful-fs/graceful-fs.js:43:10
1|mm       |     at FSReqWrap.oncomplete (fs.js:112:15)

paviro, 2017-06-11:
This is not related to the core system. Please head to the forum for further questions.

fewieden, 2017-06-11:
looks like a duplicate of https://github.com/fewieden/MMM-voice/issues/12

closed 2017-06-18 by ezar