MagicMirrorOrg/MagicMirror #910: Clock: showWeek does not honor language setting

wbleek, 2017-06-04:
When switching week display on in the Clock module (showWeek: true) the word "Week" displays even though German / de is set as language and all other modules use German.

When submitting a new issue, please supply the following information:

**Platform**: Raspberry Pi 3, Raspberrian Jessie Light.

**Node Version**: v6.10.3

**MagicMirror Version**: 2.1.1 (2017-04-01).

QNimbus, 2017-06-05:
This has been fixed in the 'development' branch, so it seems. I was only able to reproduce your issue on 'master'.

roramirez, 2017-06-06:
Hi @wbleek 

Please, close this issue.

wbleek, 2017-06-11:
I took a look at the current implementation and in my opinion it does not work for German.

English: "Week 32"
German: "32. Kalenderwoche", "32. Woche", "32. KW"

It is not idiomatic to write "Woche 32". You would use an ordinal number and put it at the start.

In rare instances "KW 32" would be acceptable (e.g. not enough space).

roramirez, 2017-06-19:
Hi @wbleek 

Please send a Pull Request with the correction for that.

Thanks

QNimbus, 2017-06-26:
If believe this can be fixed by using a templated translation string (`de.json`) and making a minor correction to the clock.js module where the translation is invoked. Of course this means that all the other translation files will need to have the same templated string for `WEEK` changed.

```
// de.json
"WEEK": "{weekNumber}. Woche",
```

```
// clock.js
weekWrapper.innerHTML = this.translate("WEEK", { weekNumber: now.week() });
```

wbleek, 2017-06-28:
Thanks for the suggestion. I will work on the implementation and send a pull request.

wbleek, 2017-09-28:
Hi, I have made the changes but for some languages I really cannot provide the right text. Should I put the english version into those translation files?
By the way: In many translation files there doesn't exist an entry for WEEK.

MichMich, 2017-09-28:
Just leave out the translations you cant add. It will fall back to English. Which is probably why WEEK is missing: it’s the same word in a lot of other languages. 

Thanks!

MichMich, 2017-10-03:
@wbleek Can I close this issue?

wbleek, 2017-10-03:
@MichMich 
Well, I did the refactoring locally on a branch. I would love to upload that branch and make a pull request. However, I get "remote: Permission to MichMich/MagicMirror.git denied to wbleek."
Do I need to get permission or am I doing it the wrong way?

nhubbard, 2017-10-03:
@wbleek You need to fork (using the button in the top right corner), clone that repository, apply your changes and push to there. Then you create a pull request to the main here.

nhubbard, 2017-10-03:
If you need help, go ahead and ask... we love first-time contributors!

wbleek, 2017-10-03:
Thanks for the help. I think I managed. First time is always the hardest. If I am successful, maybe I contribute more ;-)

closed 2017-10-03 by wbleek

Cubixzno1, 2019-03-26:
I still have a similar problem? I cant get it to write "uge" in danish. Everything else works like a charm.. any suggestions?