MagicMirrorOrg/MagicMirror #914: Zoom-Error

Moggels, 2017-06-08:
Hi,
I have some issues with the zoom-function in the config.txt.
I set the zoom for testing to a value of 0.7 like this:

`zoom: 0.7,
`
After this i wanted to set it back with deleting the code-line but nothing happened. 
I tried to reset it with:
`zoom: 1.0,
`
but nothing changes. It doesnt even mather if the number is 0 or 100, the size of everything stays the same. 

It is like the problem that icekuhn mentioned in the last post here:
https://forum.magicmirror.builders/topic/1521/zoom-scaling

fewieden, 2017-06-08:
Can't reporduce your issue.

Setting `zoom: 0.25,` gives a zoom factor of 4.
By deleting this line again it's back to default.
Also when setting `zoom: 1,`.

jjedele, 2017-06-14:
We are actually having the problem. Set the zoom in config to 2.0, changed it back but it just stayed zoomed. We even used a completely new clone of MagicMirror and the problem persisted, so it might be something in the Electron cache or something.

Setting the zoom in the CSS file worked.

jukums, 2017-06-21:
Ok, got same problem…
Connecting the keyboard and hitting Ctrl+0 to reset zoom seemed to work.
Seems that zoom value is being cached somewhere…

olexs, 2017-06-22:
There is a mechanism to cache zoom values per domain in Chrome/Chromium, this might have made its way into some versions of Electron as well and be interfering with MM settings. Could you post the specific versions and system you're running? I'll try and reproduce, since I introduced the zoom feature in the first place I might as well try and fix it :)

jjedele, 2017-07-03:
In our case it looks like Electron 1.6.11 and MagicMirror 5c41e24b9927e36ba0f7039fd87b9fa5ca82e4a1

Moggels, 2017-07-07:
@jukums thx for the tip with CTRL+0. That helped :+1:

Reotch2, 2017-07-12:
@FREDSWG in the post above, recommended updating the MagicMirror/css/custom.css file with the following.  

 body {
 	zoom: 75%;
 }

It's working great for me until the cache issue gets fixed.

DanielWeeber, 2017-08-17:
Same problem here!

hallo02, 2017-09-15:
Same Problem here!

MichMich, 2017-09-29:
This unfortunately is a Electon issue. Not much we can do.

closed 2017-09-29 by MichMich

aandrew7, 2018-01-29:
Reboot

fiveseven808, 2018-06-21:
I have this same issue. Considering 9 months has passed, maybe electron will be updated? as per the post, use an attached keyboard to my pi, I am able to do a ctrl+0 to return size to 100% however I am unable to adjust the size via configuration file afterwards. 

I may try the CSS modification as well since right now the config modification will not let me change things a second time.

MichMich, 2018-06-21:
The next version of mm will include electron 2. You might wanna check if that solves the issue.

justjim1220, 2018-06-23:
put it in your **_custom.css_** file...
`body {
    zoom: 75%;             **<<<========**** Just like this...******
    margin: 20px;
    position: absolute;
    height: calc(100% - 40px);
    width: calc(100% - 40px);
    background: #000;
    color: red;
    font-family: "Segoe Print", sans-serif;
    font-weight: 600;
    font-size: 1em;
    line-height: 1.5em;
    -webkit-font-smoothing: antialiased;
}
`

justjim1220, 2018-06-23:
this is without the zoom...

[image]

This is with zoom...

[image]

RachitSharma123, 2018-09-22:
Go to your magic mirror folder and then to css and then towards the custom.css
Inside 
body 
{
zoom: 144%;
} 
Thank me later.

l3d00m, 2021-09-22:
To summarise: 

1. Do not use the `zoom` config option. 
2. If you have used it, press CTRL+0 and it'll reset the zoom level to the original.
3. After that you can modify the zoom level inside `css/custom.css` like this
```css
body {
    zoom: 75%;
}
```

githubedu, 2024-04-29:
When running without keyboard that solved the problem of changing zoom parameter in config.js, but not getting it changed.

```
rm -rf ~/.config/Electron/
pm2 restart MagicMirror
```