MagicMirrorOrg/MagicMirror #917: Black Screen + Error launch MM.... No issue on web

kenjix95, 2017-06-12:
Hi, 

i'm on an Orange PI, i follow all instruction with this link : https://github.com/MichMich/MagicMirror/tree/develop

I launch MM² in the directory with npm start, but juste a Black Screen appear and nothing else...

i saw on the web maybe its the Version of MM², i try to downgrade Node & Npm version with 
`

Changing node version

sudo npm cache clean -f
sudo npm install -g n
sudo n 5.1.0

Changing npm version

npm install -g npm@2.4.0
`

`
Starting MagicMirror: v2.1.1
Loading config ...
Loading module helpers ...
No helper found for module: alert.
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: clock.
Initializing new module helper ...
Module helper loaded: calendar
No helper found for module: compliments.
No helper found for module: currentweather.
No helper found for module: weatherforecast.
Initializing new module helper ...
Module helper loaded: newsfeed
All module helpers loaded.
Starting server on port 8080 ...
Starting server op port 8080 ...
Server started ...
Connecting socket for: updatenotification
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: newsfeed
Starting module: newsfeed
Sockets connected & modules started ...
Whoops! There was an uncaught exception...
{ Error: listen EADDRINUSE :::8080
    at Object.exports._errnoException (util.js:1022:11)
    at exports._exceptionWithHostPort (util.js:1045:20)
    at Server._listen2 (net.js:1262:14)
    at listen (net.js:1298:10)
    at Server.listen (net.js:1376:9)
    at new Server (/root/MagicMirror/js/server.js:27:9)
    at /root/MagicMirror/js/app.js:219:18
    at loadNextModule (/root/MagicMirror/js/app.js:168:5)
    at /root/MagicMirror/js/app.js:163:6
    at Class.loaded (/root/MagicMirror/modules/node_modules/node_helper/index.js:19:3)
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '::',
  port: 8080 }
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
Launching application.

MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
`

If somebody have issue or soluce

kenjix95, 2017-06-12:
i solve this

cd MagicMirror/vendor
npm install

i reboot my OPI

a run MagicMirror and TADAM ! 

but it's in english ^^

I go to find some French module

rejas, 2017-06-12:
FYI: if the error logs say something like this: 
`{ Error: listen EADDRINUSE :::8080`
be sure that no other instance of MagicMirror (or any other program) is already running and blocking the port 8080. Thats why rebooting might have solved it for you.

kenjix95, 2017-06-12:
i've clean my SD Card,

just i have install vncserver

closed 2017-06-21 by MichMich