MagicMirrorOrg/MagicMirror #922: Feature: Allow for server to be run remotely instead of on localhost

QNimbus, 2017-06-18:
As the title says. I would like to make some changes to allow for the node server to be run on a different server/network location than the electron client. Specifying the server ip/hostname would be a configurable option with a default of 'localhost'.

I would be more than happy to create a PR for this feature if the community supports this feature

MichMich, 2017-06-21:
It might be a good idea to investigate the amount of work that needs to be done before you start this overhaul. If this is doable with a few minor changes, feel free to send a PR.

roramirez, 2017-06-21:
Sound good for me +1. Today you can set the address where the server can start. Only I see the part is missing is say to client app (electron) address where to get the data, maybe ws could be hard part of this work.

QNimbus, 2017-06-23:
It will be more work than I had anticipated because the client and server are more intertwined than I had originally thought. However it is possible and I would still like to have a shot at doing this if everyone is on board with this idea/feature.

To get the config from the server to the client (and to prevent having two separate config files) I found a really neat way of using a cookie (that gets removed immediately) to get the config from the server node application to the electron client. Inspired by the [NPM clientconfig package](https://github.com/henrikjoreteg/clientconfig).

I will try to submit a PR as soon as possible but I have quite a few other responsibilities that keep eating away at the time I can invest in this.

QNimbus, 2017-07-27:
I will try to write  some automated tests for this implementation - the feature itself still needs to be tested by users/developers since it's now in the `develop` branch.

Can I close this issue or do we still need this open since it hasn't been properly tested yet?

roramirez, 2017-07-28:
The ideal to me is close this after wrote the tests. Because a new feature with automated tests we be sure will work like as we expected.

MichMich, 2017-09-29:
Closed due inactivity (and since it probably already works on the dev branch :))

closed 2017-09-29 by MichMich