MagicMirrorOrg/MagicMirror #925: Perpetually growing array in default Compliments module

talkinnsmak, 2017-06-26:
**Platform**: Raspberry Pi 3, Raspbian Jessie

**Node Version**: v6.11.0

**MagicMirror Version**: v2

**Description**: In compliments.js, the complimentArray function:
At the top of the function, var compliments is declared, then assigned to a compliment array based on one of the time of day categories, morning/afternoon/evening. Further down, the "anytime" compliment array is pushed onto the back of compliments. However, by using:
compliments = this.config.compliments.<morning/afternoon/evening> 
it appears to be assigning the original memory address of the default array to the compliments variable. So when the anytime array is later pushed onto it, it's pushing it onto the original default array in memory. Any subsequent calls to the function will assign to the compliments reference the array related to morning/afternoon/evening which now also has the anytime array appended to it from the previous iteration, and then appends the anytime array again onto the end of it. This continually grows the array with duplicates of the anytime array with every update/function call and slowly, eventually, you'll run out of memory.

**Steps to Reproduce**: To see the evidence of it, In the randomCompliment function further down, locate the line:
return compliments[index];
and change it to:
return (compliments[index] + " " + compliments.length;

**Expected Results**: During the very first compliment displayed on launch, you'll see the screen output the value of the combined sizes of the anytime array and the array related to whatever time of day category you happen to be testing this in next to the randomly selected compliment (provided compliments are present in all the default list categories at the top). This number should remain the same every time it cycles a new compliment as long as its happening in the same time of day category (morning/afternoon/evening).

**Actual Results**: Instead, you'll notice that the value of compliments.length increases by a value equal to the number of compliments in the default anytime array. This value keeps increasing every time a new compliment is displayed. After a long while, you'll only start to see compliments from the anytime list as all the duplicates begin to dominate the choices.

**Configuration**: N/A

**Additional Notes**: I seem to have fixed the issue by assigning compliments to a new array at the top and then pushing the morning/afternoon/evening array to it rather than directly assigning it. This forces a new memory address for the compliment array and avoids appending to the original default arrays. 

Apologies for the somewhat long-winded description, I've not been programming for very long. It's a small, short fix, but I'm happy to take a stab at doing a PR with the change I went with, if needed, unless there's a better fix I'm not aware of.

QNimbus, 2017-06-26:
For what it's worth, I was able to reproduce the issue you address in the 'master' branch. In the 'develop' branch I'm not able to reproduce it. I will check the changes to the `compliments` module to confirm why or how this was fixed.

On a related note - maybe it's time to release a new version because there have been quite a few issues reported recently that seem to occur in the 'master' (release) branch that have been fixed since. Last release was early April so maybe we're due for another release?

QNimbus, 2017-07-02:
@talkinnsmak Can you confirm that this issue is resolved in the new release?

MichMich, 2017-09-29:
Fixed in the next release.

closed 2017-09-29 by MichMich