MagicMirrorOrg/MagicMirror #932: npm start issue

BrezhDev, 2017-07-03:
**Platform**: Raspberry 2

**Node Version**: 6.11

**MagicMirror Version**: Last one (07/01)

**Description**: 
>DISPLAY=:0 npm start
magicmirror@2.1.2 start /home/pi/MagicMirror
sh run-start.sh

>npm ERR! Linux 4.9.28-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! node v6.11.0
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.1.2 start: `sh run-start.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the magicmirror@2.1.2 start script 'sh run-start.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     sh run-start.sh
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

>npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log

rejas, 2017-07-03:
Could you post your npm-debug.log (if it exists in your MagicMirror directory)?

stefan-hoffmann, 2017-07-10:
Hello,

i have the same Problem. Want to setup complete new and running into this error.
Running on RPI3 with latest Raspbian Lite -> upgraded to PIXEL GUI.
No Errors at Setup via Automatic Installer...

My Log:

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/nodejs', '/usr/bin/npm', 'start' ]
2 info using npm@3.10.10
3 info using node@v6.11.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.1.2~prestart: magicmirror@2.1.2
6 silly lifecycle magicmirror@2.1.2~prestart: no script for prestart, continuing
7 info lifecycle magicmirror@2.1.2~start: magicmirror@2.1.2
8 verbose lifecycle magicmirror@2.1.2~start: unsafe-perm in lifecycle true
9 verbose lifecycle magicmirror@2.1.2~start: PATH: /usr/lib/node_modules/npm/bin/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
10 verbose lifecycle magicmirror@2.1.2~start: CWD: /home/pi/MagicMirror
11 silly lifecycle magicmirror@2.1.2~start: Args: [ '-c', 'sh run-start.sh' ]
12 info lifecycle magicmirror@2.1.2~start: Failed to exec start script
13 verbose stack Error: magicmirror@2.1.2 start: `sh run-start.sh`
13 verbose stack spawn ENOENT
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/lib/utils/spawn.js:33:16)
13 verbose stack     at emitTwo (events.js:106:13)
13 verbose stack     at ChildProcess.emit (events.js:191:7)
13 verbose stack     at maybeClose (internal/child_process.js:891:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
14 verbose pkgid magicmirror@2.1.2
15 verbose cwd /home/pi/MagicMirror
16 error Linux 4.9.36-v7+
17 error argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
18 error node v6.11.0
19 error npm  v3.10.10
20 error file sh
21 error code ELIFECYCLE
22 error errno ENOENT
23 error syscall spawn
24 error magicmirror@2.1.2 start: `sh run-start.sh`
24 error spawn ENOENT
25 error Failed at the magicmirror@2.1.2 start script 'sh run-start.sh'.
25 error Make sure you have the latest version of node.js and npm installed.
25 error If you do, this is most likely a problem with the magicmirror package,
25 error not with npm itself.
25 error Tell the author that this fails on your system:
25 error     sh run-start.sh
25 error You can get information on how to open an issue for this project with:
25 error     npm bugs magicmirror
25 error Or if that isn't available, you can get their info via:
25 error     npm owner ls magicmirror
25 error There is likely additional logging output above.
26 verbose exit [ 1, true ]

stefan-hoffmann, 2017-07-10:
Statement: No idea why some text in the log is crossing out, must be a copy/paste problem...

rorypwj, 2017-07-11:
Hey, I'm having the same problem same errors, etc. Any luck on solving this?

MichMich, 2017-09-29:
Closed due to inactivity.

closed 2017-09-29 by MichMich

Y0nnyy, 2017-12-12:
Why is this closed? Has the issue been resolved? Currently i have the same problem.
```npm start``` running in terminal on the device is working, but ```DISPLAY=:0 npm start``` over ssh does not.

MichMich, 2017-12-12:
@Y0nnyy Do you see any errors?

Y0nnyy, 2017-12-12:
Yes: 
```
> magicmirror@2.1.3 start /home/hcm/MagicMirror
> sh run-start.sh

No protocol specified

npm ERR! Linux 4.9.65-v7+
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "start"
npm ERR! node v6.12.0
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.1.3 start: `sh run-start.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the magicmirror@2.1.3 start script 'sh run-start.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     sh run-start.sh
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/hcm/MagicMirror/npm-debug.log
``` 

And the log file: 
```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@3.10.10
3 info using node@v6.12.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.1.3~prestart: magicmirror@2.1.3
6 silly lifecycle magicmirror@2.1.3~prestart: no script for prestart, continuing
7 info lifecycle magicmirror@2.1.3~start: magicmirror@2.1.3
8 verbose lifecycle magicmirror@2.1.3~start: unsafe-perm in lifecycle true
9 verbose lifecycle magicmirror@2.1.3~start: PATH: /usr/lib/node_modules/npm/bin/node-gyp-bin:/home/hcm/MagicMirror/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
10 verbose lifecycle magicmirror@2.1.3~start: CWD: /home/hcm/MagicMirror
11 silly lifecycle magicmirror@2.1.3~start: Args: [ '-c', 'sh run-start.sh' ]
12 silly lifecycle magicmirror@2.1.3~start: Returned: code: 1  signal: null
13 info lifecycle magicmirror@2.1.3~start: Failed to exec start script
14 verbose stack Error: magicmirror@2.1.3 start: `sh run-start.sh`
14 verbose stack Exit status 1
14 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/lib/utils/lifecycle.js:255:16)
14 verbose stack     at emitTwo (events.js:106:13)
14 verbose stack     at EventEmitter.emit (events.js:191:7)
14 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/lib/utils/spawn.js:40:14)
14 verbose stack     at emitTwo (events.js:106:13)
14 verbose stack     at ChildProcess.emit (events.js:191:7)
14 verbose stack     at maybeClose (internal/child_process.js:920:16)
14 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:230:5)
15 verbose pkgid magicmirror@2.1.3
16 verbose cwd /home/hcm/MagicMirror
17 error Linux 4.9.65-v7+
18 error argv "/usr/bin/node" "/usr/bin/npm" "start"
19 error node v6.12.0
20 error npm  v3.10.10
21 error code ELIFECYCLE
22 error magicmirror@2.1.3 start: `sh run-start.sh`
22 error Exit status 1
23 error Failed at the magicmirror@2.1.3 start script 'sh run-start.sh'.
23 error Make sure you have the latest version of node.js and npm installed.
23 error If you do, this is most likely a problem with the magicmirror package,
23 error not with npm itself.
23 error Tell the author that this fails on your system:
23 error     sh run-start.sh
23 error You can get information on how to open an issue for this project with:
23 error     npm bugs magicmirror
23 error Or if that isn't available, you can get their info via:
23 error     npm owner ls magicmirror
23 error There is likely additional logging output above.
24 verbose exit [ 1, true ]

```