MagicMirrorOrg/MagicMirror #933: Commit c15148f breaks analog clock alignment when using position other than "top_left"

shbatm, 2017-07-05:
**Platform**: RPi 3

**Node Version**: v7.10.0

**MagicMirror Version**: v2.1.2

**Description**: After updating to v2.1.2, the analog face of the clock module is stuck on the left side of the mirror when using both analog and digital face.

**Steps to Reproduce**: Using following config for clock:
```js
        {
            module: "clock",
            position: "middle_center",
            config: {
                displayType: "both"
            }
        },
```

**Expected Results**: Both digital and analog clocks to be centered in screen.

**Actual Results**: Digital clock is centered, analog clock is left-aligned

**Configuration**: See above

**Additional Notes**: Reverting changes made in commit https://github.com/MichMich/MagicMirror/commit/c15148fc07119ed772438442fae6fa9d47b51946 fixes the problem.

QNimbus, 2017-07-06:
I will have a look at this later this week when I get home; I was responsible for the commit you mentioned and I will try to fix the issue you described along with the allignment issue I tried to address originally.

roramirez, 2017-07-06:
The style of anolog clock is a mess ;).  Here there is a previous fix in a similar issue https://github.com/MichMich/MagicMirror/commit/949f7587dc5fbb8975883728872fb6d981ade33f

This maybe can help you

closed 2017-09-29 by MichMich