MagicMirrorOrg/MagicMirror #936: analog clock position went wrong after update

apm1007, 2017-07-07:
hi, today i made an update to the latest version. i use the clock in top_bar. After the update the analog clock jumped to the left side. Any ideas how to fix this? Dunno where there error comes from.
[image]
[image]

fewieden, 2017-07-07:
probably related to #933

QNimbus, 2017-07-07:
This is addressed in [Issue 993](https://github.com/MichMich/MagicMirror/issues/933).

QNimbus, 2017-07-08:
You can probably close this issue since [issue #933](https://github.com/MichMich/MagicMirror/issues/933) already addresses this problem.

closed 2017-09-29 by MichMich