MagicMirrorOrg/MagicMirror #937: npm start issue

zydecat, 2017-07-07:
Can't get mirror to start up since updating to the latest version. It accepts the pm2 start mm.sh command, but wont actually start the application. Here is the debug log...

0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start', 'dev' ]
2 info using npm@3.8.3
3 info using node@v5.10.1
4 verbose config Skipping project config: /home/pi/.npmrc. (matches userconfig)
5 verbose stack Error: ENOENT: no such file or directory, open '/home/pi/package.json'
5 verbose stack     at Error (native)
6 verbose cwd /home/pi
7 error Linux 4.1.19-v7+
8 error argv "/usr/local/bin/node" "/usr/local/bin/npm" "start" "dev"
9 error node v5.10.1
10 error npm  v3.8.3
11 error path /home/pi/package.json
12 error code ENOENT
13 error errno -2
14 error syscall open
15 error enoent ENOENT: no such file or directory, open '/home/pi/package.json'
16 error enoent ENOENT: no such file or directory, open '/home/pi/package.json'
16 error enoent This is most likely not a problem with npm itself
16 error enoent and is related to npm not being able to find a file.
17 verbose exit [ -2, true ]

cheers

fewieden, 2017-07-07:
It tries to start the application in `/home/pi`, but it should look in `/home/pi/MagicMirror`

zydecat, 2017-07-08:
I think that is all working, I have checked my auto start instructions from the main page. Trouble is I run the pm2 start mm.sh command and it responds as if it worked, but the mirror doesnt start up. 
[image]

It was working until last night when I did a git pull. I'm a bit of a noob but I have updated in the past with no trouble, and there is no issue with my config when I checked.

Disappointing because I was hoping to use the compliments section for a birthday party we are holding today. Oh well. Just want to get it fixed. Let me know if there is anything else I can try...

fewieden, 2017-07-08:
Hi Mat, if you get in touch with me on the teamspeak server I might can help you in time. Or send me a pm on the forum.

TeamSpeak Magic Mirror Help @ 173.68.7.240 http://www.teamspeak.com/downloads.html

https://forum.magicmirror.builders/user/strawberry-3-141

zydecat, 2017-08-07:
Hi,
An Update. (after a busy few weeks)

I have finally had the time and got my mirror working again. From what I can gather, when I was forced to update the raspberry pi, I seem to have lost a lot of config (particularly with reference to PM2.) Has anyone else experienced mirror problems after an update to the Pi? 

The second thing with the update to the pi was that for whatever reason, the "auto installer for raspberry pi only" no longer worked (tried three or four times). Ended up following the instructions for a manual installation and this worked, Then just redid my config to reinstall PM2 etc. 

Anyway might be handy to know.

Cheers,

Mat

eomisterman, 2017-08-15:
Hi Mat,

I'm having the same issue. What exactly did you do to fix this problem? How did you redo the config to reinstall PM2 etc.?

Thanks.

roramirez, 2017-08-16:
@zydecat Can you reproduce the issue?  if not please close this issue. 

@eomisterman You can find support for this kind of troubles in https://forum.magicmirror.builders/

zydecat, 2017-08-16:
Hi @eomisterman and @roramirez,
The issues with the mirror and pm2 not running seem to have been caused by the latest firmware update to the raspberry pi itself. This seemed to remove or mess up aspects of the mirror config. In trying to rectify the problem, I removed everything and tried reinstalling using the "Raspberry Pi Only" auto installing process ([https://github.com/MichMich/MagicMirror#automatic-installer-raspberry-pi-only](url)), but found that this didn't work any longer.
I found that the mirror would only install using the Manual Installation instructions ([https://github.com/MichMich/MagicMirror#automatic-installer-raspberry-pi-only](url)) Then ran through the steps for configuring the Pi ([https://github.com/MichMich/MagicMirror/wiki/Configuring-the-Raspberry-Pi](url)) (some of the config had been saved, other config had to be re-entered) and the auto start ([https://github.com/MichMich/MagicMirror/wiki/Auto-Starting-MagicMirror](url)) (PM2 installation). 

So just saying that for new installs, particularly for recently updated Pi's, the auto installer needs to be updated or something. For now, you need to run the manual instructions.

Hope that helps.

Cheers,

Mat

MichMich, 2017-09-29:
@nhubbard suggested to close this. Feel free to reopen and kick him in the butt if he's mistaken. ;)