MagicMirrorOrg/MagicMirror #940: Calendar times incorrect since update

mc-landin, 2017-07-08:
Since the update to the latest PI version my calendar times are incorrect. The timezone is set correctly and set to 24H.

Yet, all times after noon are set 12 hours too early.

Tuxdiver, 2017-07-09:
Problem is in calendar.js:
```
diff --git a/modules/default/calendar/calendar.js b/modules/default/calendar/calendar.js
index 9ab864b9..672a47b 100644
--- a/modules/default/calendar/calendar.js
+++ b/modules/default/calendar/calendar.js
@@ -81,7 +81,7 @@ Module.register("calendar", {
                case 24: {
                        moment.updateLocale(config.language, {
                                longDateFormat: {
-                                       LT: "hh:mm"
+                                       LT: "HH:mm"
                                }
                        });
                        break;
```

should fix the problem

QNimbus, 2017-07-09:
You are correct - was sloppy testing on my behalf. The PR should fix this problem. Thanks for reporting!

roramirez, 2017-07-20:
@MichMich Please, close this issue :). Thanks @QNimbus

closed 2017-07-21 by mc-landin