MagicMirrorOrg/MagicMirror #942: error with config.js

abacon118, 2017-07-09:
I just installed magic mirror and opened index.html and it told me that there was an error with the conifg.js file.  I re-named the file and moved it and still had no luck.  What should i do?

fewieden, 2017-07-09:
you have to start the server, you can't just open the html file.

QNimbus, 2017-07-09:
`npm start` should do the trick when run in your MagicMirror folder. Or you can run `node serveronly` and use a browser to open the MagicMirror site.

Let us know if this solved your problem - if it's resolved, please close this issue

abacon118, 2017-07-09:
When i run npm start from the magic mirror directory i get this.

[image]

rejas, 2017-07-09:
So have you node installed? What do you get when you run "node -v"-command in that directory?

QNimbus, 2017-07-09:
Maybe you could try upgrading NodeJS to the latest version?

`$ sudo apt update`
`$ sudo apt install nodejs`

roramirez, 2017-07-11:
If are you using the manual installation please follow the steps
https://github.com/MichMich/MagicMirror#manual-installation

After clone the repository
```
cd ~/MagicMirror
npm install
npm start
```

If not, you can use the automatic installer script
`bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh)"`

Good luck!

MichMich, 2017-09-29:
Please refer to http://forum.magicmirror.builders for support and troubleshooting.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2017-09-29 by MichMich

FastGeneration, 2019-04-17:
Hy sir please tell me about clone error in installing smart mirror in raspberry
Please tell me solution thank u