MagicMirrorOrg/MagicMirror #944: installer error

cfdresearch, 2017-07-09:
Try to install MagicMirror using installer and got a bash error (below):

` curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh | bash`

`[PM2] Spawning PM2 daemon with pm2_home=/home/pi/.pm2
[PM2] PM2 Successfully daemonized
[PM2][WARN] Applications MagicMirror not running, starting...
[PM2] App [MagicMirror] launched (1 instances)
┌─────────────┬────┬──────┬──────┬────────┬─────────┬────────┬─────┬──────────┬──────────┐
│ App name    │ id │ mode │ pid  │ status │ restart │ uptime │ cpu │ mem      │ watching │
├─────────────┼────┼──────┼──────┼────────┼─────────┼────────┼─────┼──────────┼──────────┤
│ MagicMirror │ 0  │ fork │ 2202 │ online │ 0       │ 0s     │ 0%  │ 2.3 MB   │ enabled  │
└─────────────┴────┴──────┴──────┴────────┴─────────┴────────┴─────┴──────────┴──────────┘
 Use `pm2 show <id|name>` to get more details about an app
[PM2] Saving current process list...
[PM2] Successfully saved in /home/pi/.pm2/dump.pm2
bash: line 157: syntax error near unexpected token `fi'
bash: line 157: `fi'`

`pi@mmirror2pi:~ $ uname -a
Linux mmirror2pi 4.9.36-v7+ #1015 SMP Thu Jul 6 16:14:20 BST 2017 armv7l GNU/Linux`

`pi@mmirror2pi:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        30G  3.4G   25G  12% /
devtmpfs        458M     0  458M   0% /dev
tmpfs           462M  7.3M  455M   2% /dev/shm
tmpfs           462M  6.4M  456M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           462M     0  462M   0% /sys/fs/cgroup
/dev/mmcblk0p1   42M   21M   21M  51% /boot
tmpfs            93M     0   93M   0% /run/user/1000`

fewieden, 2017-07-09:
This command is deprecated. Use `bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh)"` instead

cfdresearch, 2017-07-09:
Thanks a lot, now it's fine. I just used this tutorial: [https://forum.magicmirror.builders/topic/236/complete-setup-tutorial/6](url). Issue may be closed.

closed 2017-07-09 by cfdresearch

Jopyth, 2017-07-14:
I fixed the link in the mentioned tutorial.