MagicMirrorOrg/MagicMirror #947: Powering off monitor, cant connect to X server when powering back ON

ImTheBadWolf, 2017-07-15:
Raspberry pi zero, midori browser

node - 4.2.1

I made a script to power on and off monitor so its not always on. The problem is that when I turn it off and leave it off for a while it then wont turn on again. It gives me error "connection to X server lost"

In script to power off the monitor I have:  "tvservice -o"
And in script for powering it back on I have this:  "tvservice -p" and "xset -display :0 dpms force on"

It works well but when its turned off for a longer period then it wont load the mirror, only black screen appears and when I press ctrl + w to terminate the midori browser it says "connection to X server lost"
[image]

shbatm, 2017-07-16:
Try using `tvservice --preferred && sudo chvt 6 && sudo chvt 7` as the command to power back on.

ImTheBadWolf, 2017-07-16:
Thank you for help but its not working either

roramirez, 2017-07-20:
For this kind of questions please to [referred on the forum](https://forum.magicmirror.builders/). This is better place to kind of support.