MagicMirrorOrg/MagicMirror #948: Sharing node_helper resources across multiple mirrors

shbatm, 2017-07-16:
**Not a bug: possible feature request**

I started toying with the ability to share a Module's node_helper and socket across multiple instances and devices.  I wanted to share what I've done in case anyone else finds it useful or thinks it could/should be incorporated. 

Working code preview is here: https://github.com/shbatm/MagicMirror/commit/ccd958f203636606218e8101c573582397c1fc59 

**Usage:**
As simple as I could make it: To connect a module to a remote server's socket/node_helper, add `remote: "http://othermirrorurl:port/",` under the module's definition in your config.js file. For Example:
````js
{ module: "calendar",
  header: "US Holidays",
  position: "top_left",
  remote: "http://localhost:8080",
  config: { ...
````
Sample config file to share weather and newsfeed modules: https://gist.github.com/shbatm/ccd5de1db3c6f8cc22843d784e2c77b7

**Why?**
 * Offload resource intensive modules to a server with more power, but still allow for local node modules to control things like the local screen power or GPIO devices.
 * Reduce API calls when using multiple mirrors
 * Run multiple instances of MM with different config files (for different clients) on the same server, but limit the resource use by not creating multiple instances of the node_helpers when shared between servers.  

Easier to explain with a picture:
[image]

**Notes:**
 * Does not create a new node_helper instance on local device when using a remote client
 * Does not require changes to each module's code
 * Ability to run multiple instances with different configs, but share helpers (e.g. using `MM_CONFIG_FILE=config/config_XXX.js node serveronly` with a different port for server set in each config file).  
     * Example config for a second instance with shared news/calendar modules is linked above.

fewieden, 2017-07-16:
That's basically what I did three months ago https://github.com/fewieden/MMM-syslog-slave to connect instances to syslog master.

roramirez, 2017-07-20:
I like the idea but @MichMich have it the last word of this

MichMich, 2017-09-29:
Too much of an edge case to do such of an overhaul. I like @fewieden's solution: implementing this feature in a module.

I close this for now. Feel free to reopen is nessecery.

closed 2017-09-29 by MichMich