MagicMirrorOrg/MagicMirror #950: Issue: Empty ipWhitelist array does not allow any host to access MM

QNimbus, 2017-07-17:
When you set `ipWhitelist = []` in the config file, it should allow access from everyone to the MM instance according to the documentation. Instead with an empty whitelist (`express-whitelist` module) it doesn't allow any host to access the MM instance.

I think this could be solved by replacing the following line in `server.js` 

`var result = ipfilter(config.ipWhitelist, {mode: "allow", log: false})`

with

`var result = ipfilter(config.ipWhitelist, {mode: config.ipWhitelist.length === 0 ? "deny" : "allow", log: false})` 

This will result in `ipWhitelist = []` in the config file to allow any and all hosts to access the MM instance.

QNimbus, 2017-07-17:
To prevent some confusion during configuration (users customizing their `config.js` file), I've changed the default `address` which is used to specify addresses/interfaces where the server listens on, to listen on all interfaces by default. 

If a user changes the default ipWhitelist (which by default only allows localhost) he/she won't be hampered by the fact that they might not have the `address` property configured in their local config.

qistoph, 2017-07-20:
From a security perspective I would opt to prevent users from accidentally opening up access to their mirror. Listening only on, or allowing access from localhost woudl be wise from that perspective.

Take a look at [Mirai](https://en.wikipedia.org/wiki/Mirai_(malware)) for example.

Raspberry did a major change regarding these issues, by [disabling SSH by default](https://www.raspberrypi.org/blog/a-security-update-for-raspbian-pixel/).

Therefore I would suggest to keep the default to limited access.

QNimbus, 2017-07-20:
I agree with you - I will try to come up with a more elegant solution, and send a PR with reference to this issue.

roramirez, 2017-08-22:
There a log when is open a full ipWhitelist.

qistoph, 2017-08-25:
@QNimbus, any updates?

@roramirez, what do you mean?

roramirez, 2017-08-25:
When you are using a full IpWhitelist, MM alert you with a  warning log entry in console.
https://github.com/MichMich/MagicMirror/blob/develop/js/server.js#L30

qistoph, 2017-08-25:
My concern is that the default install & run will leave you with an application listening on all network devices on any IP (include IPv6). Given the increase in IoT devices being abused I think this is something to try and prevent as software developers.

So I think the default install & run should at least prevent one of the two; 

- listen only on the localhost interface, or 
- listen only on the localhost IPs 127.0.0.1 and ::1

roramirez, 2017-08-25:
I've done the test and the log is show it.

```
root@testing-rpi-iso:~/MagicMirror# grep white config/config.js
root@testing-rpi-iso:~/MagicMirror# grep White config/config.js
	ipWhitelist: [],
root@testing-rpi-iso:~/MagicMirror# node serveronly/
Starting MagicMirror: v2.1.3-dev
Loading config ...
Loading module helpers ...
No helper found for module: alert.
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: clock.
Initializing new module helper ...
Module helper loaded: calendar
No helper found for module: compliments.
No helper found for module: currentweather.
No helper found for module: weatherforecast.
Initializing new module helper ...
Module helper loaded: newsfeed
All module helpers loaded.
Starting server on port 8080 ... 
You're using a full whitelist configuration to allow for all IPs
Server started ...
Connecting socket for: updatenotification
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: newsfeed
Starting module: newsfeed
Sockets connected & modules started ...

Ready to go! Please point your browser to: http://localhost:8080
```

MichMich, 2017-09-29:
Fixed in next release.

closed 2017-09-29 by MichMich

mstreicher, 2020-06-17:
This is already not fixed!