MagicMirrorOrg/MagicMirror #958: Newsfeed error

FranciscoGomez90, 2017-07-25:
I just installed Magic Mirror and I¡m fooling around with the modules. The sample config give me this error when loading newsfeed.

```
Starting MagicMirror: v2.1.2
Loading config ...
Loading module helpers ...
No helper found for module: alert.
No helper found for module: clock.
No helper found for module: compliments.
WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module 'node_helper'
Loading module helpers ...
No helper found for module: alert.
No helper found for module: clock.
No helper found for module: compliments.
App threw an error during load
Error: Cannot find module 'node_helper'
    at Module._resolveFilename (module.js:470:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._load (module.js:418:25)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/newsfeed/node_helper.js:8:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/newsfeed/node_helper.js:86:3)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32)
Whoops! There was an uncaught exception...
{ Error: Cannot find module 'node_helper'
    at Module._resolveFilename (module.js:470:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._load (module.js:418:25)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/newsfeed/node_helper.js:8:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/newsfeed/node_helper.js:86:3)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32) code: 'MODULE_NOT_FOUND' }
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
Launching application.
```

My config looks like  this

```
modules: [
                {
                        module: "alert",
                },
                {
                        module: "clock",
                        position: "top_left"
                },
                {
                        module: "compliments",
                        position: "lower_third"
                },
                {
                        module: "newsfeed",
                        position: "bottom_bar", // This can be any of the regions. Best results in center regions.
                        config: {

                                feeds: [
                                        {
                                                title: "New York Times",
                                                url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml",
                                        },
                                        {
                                                title: "BBC",
                                                url: "http://feeds.bbci.co.uk/news/video_and_audio/news_front_page/rss.xml?edition=uk",
                                        },
                                ]
                        }
                }
        ]
```

Any idea why I get this error? If I remove the nodule:newsleef section everything is working properly.
Besides, I got a similar error when loading the calenda module.

Thank you

cowboysdude, 2017-07-26:
Yes it looks like your node_helper file for the news is missing.  Can you check the modules/default/newsfeed directory has a node_helper in it.

FranciscoGomez90, 2017-07-26:
Yes. It has node_helper.js :S

FranciscoGomez90, 2017-07-26:
Nevermind, I just reinstalled Magic Mirror and everything worked just fine :S

roramirez, 2017-07-27:
@FranciscoGomez90 Maybe there was missing something `npm install`. Please close the issue.

closed 2017-07-27 by FranciscoGomez90

XaviArnaus, 2018-12-27:
Just for further info: If you're developing a new module, but your code is anywhere else in your disk rather than in the `modules` folder, a simple `ln -s the/path/to/your/module` symbolic link from the `modules` folder doesn't work, producing the same error message. The module needs to be **physically** placed (or cloned) into `modules`

alexlao, 2019-01-30:
@XaviArnaus Why's that? Are there any alternatives you know I could go about as opposed to always copying my module into the directory? I want to start a Git repo for my own module and work on it in its standalone directory -- cloning the module every time is a bit tedious. Thanks for your input though, I'm glad I found your response. I spent about an hour trying to get the symlink to work until I found this thread.

XaviArnaus, 2019-04-08:
I did not do any further investigation of the cause. I understood that modules should be cloned into the `modules` folder to work properly. Very uncomfortable. What I do is to maintain my own module in a separate repo, clone it inside `modules`, and develop from there. If you find an alternative I would like to know :-)