MagicMirrorOrg/MagicMirror #960: every time i try to run it does not work, this is what i get every time i npm start

laffey2457, 2017-07-27:
pi@raspberrypi:~/MagicMirror $ npm start

> magicmirror@2.1.2 start /home/pi/MagicMirror
> sh run-start.sh

/home/pi/MagicMirror/node_modules/electron/index.js:9
  throw new Error('Electron failed to install correctly, please delete node_modules/electron and try installing again')
  ^

Error: Electron failed to install correctly, please delete node_modules/electron and try installing again
    at Object.<anonymous> (/home/pi/MagicMirror/node_modules/electron/index.js:9:9)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/MagicMirror/node_modules/electron/cli.js:3:16)
    at Module._compile (module.js:570:32)

npm ERR! Linux 4.9.28-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! node v6.11.1
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.1.2 start: `sh run-start.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the magicmirror@2.1.2 start script 'sh run-start.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     sh run-start.sh
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log
pi@raspberrypi:~/MagicMirror $

QNimbus, 2017-07-27:
I would suggest deleting the `node_modules` folder inside your `MagicMirror` folder and running `npm install` again.

Can you let us know if this solves your issue?

willfri, 2017-07-27:
I got the same error. Installed MM on a clean Raspbian installation.
So I removed `node_modules/electron` and ran `npm install` again. After this I got an new error:

```
root@raspberrypi:~/MagicMirror# npm install

> electron@1.6.11 postinstall /root/MagicMirror/node_modules/electron
> node install.js

module.js:471
    throw err;
    ^

Error: Cannot find module '/root/MagicMirror/node_modules/electron/install.js'
    at Function.Module._resolveFilename (module.js:469:15)
    at Function.Module._load (module.js:417:25)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:389:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:504:3
npm WARN stylelint-config-recommended@1.0.0 requires a peer of stylelint@^8.0.0                                                        but none was installed.
npm WARN stylelint-config-standard@17.0.0 requires a peer of stylelint@^8.0.0 bu                                                       t none was installed.
npm ERR! Linux 4.9.35-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! node v6.11.1
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! electron@1.6.11 postinstall: `node install.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the electron@1.6.11 postinstall script 'node install.js'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the electron package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node install.js
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs electron
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls electron
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /root/MagicMirror/npm-debug.log
```
Due to the new error I found this [issue](https://github.com/electron-userland/electron-prebuilt/issues/263).
After I executed the command from the above issue, I tried `npm start` again and got this:
```
> magicmirror@2.1.2 start /root/MagicMirror
> sh run-start.sh

No protocol specified

npm ERR! Linux 4.9.35-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! node v6.11.1
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.1.2 start: `sh run-start.sh`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the magicmirror@2.1.2 start script 'sh run-start.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     sh run-start.sh
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /root/MagicMirror/npm-debug.log
```

fewieden, 2017-07-28:
you did some fancy sudo stuff `/root/MagicMirror`.

I recommend to delete the whole magicmirror directory and install the project as pi user to prevent access right issues in the future

open a terminal go to `cd ~` and execute `bash -c "$(curl -sL https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh)"`

willfri, 2017-07-28:
Thank you, now it works. I tried root because on an old Raspbian I got a error with pi user yesterday.

closed 2017-09-29 by MichMich