MagicMirrorOrg/MagicMirror #976: Issue: 12-hour clock format incorrect after midnight

QNimbus, 2017-07-31:
**Platform**: Windows 10

**Node Version**: v8.2.1

**MagicMirror Version**: v2.1.3-dev

**Description**: When using 12-hour clock format, 00:30 is displayed as 12:30 AM

**Steps to Reproduce**: Set computer clock to somewhere between midnight and 01:00 (AM) and make sure that MM uses the 12-hour clock format to display the time

**Expected Results**: When the time is (for example) 00:25 AM it should display 00:25 AM

**Actual Results**: At 00:25 AM the clock shows 12:25 AM

**Configuration**: 
```
var config = {
	port: 8080,
	// ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"], // Set [] to allow all IP addresses.
	ipWhitelist: [],

	language: "nl",
	timeFormat: 12,
	units: "metric",
```

QNimbus, 2017-08-01:
I just realized after some 'Google' research that while 12:30 AM is technically correct, it is a bit ambiguous. So I will close this issue since the clock seems to be working correctly - I guess it underlines my preference for the 24-hour clock once more ;)

closed 2017-08-01 by QNimbus

roramirez, 2017-08-02:
Maybe create a poll for this cases for our users.