MagicMirrorOrg/MagicMirror #980: 2.1.2 calendar not displaying full day events

cschwarz007, 2017-08-04:
2.1.2 calendar does not display any full day events. The error can be reproduced on my RaspberryPi with a full day event in either calendar I am including in config.js:

`		{
			module: 'calendar',
			position: 'top_left',
			config: {
				maximumEntries: 20,
				fadePoint: 0.7,
				maxTitleLength: 27,
				calendars: [
					{
						symbol: 'calendar-plus-o',
						url: 'https://calendar.google.com/calendar/ical/[removed]/basic.ics'
					},
					{
						symbol: 'birthday-cake',
			            url: 'https://calendar.google.com/calendar/ical/[removed]/basic.ics'
					}
				],
				titleReplace: {
					"'s Birthday": ""
				},
			}
		},`

roramirez, 2017-08-05:
In the other release it is working?

Can you testing in 2.1.1?

cschwarz007, 2017-08-06:
Yes, it used to work in 2.1.1 and 2.1.0

roramirez, 2017-08-07:
This can be related with #926

@QNimbus, can you check?

BLoodConstructions, 2017-08-15:
Is ist possible to config the calendar that all events from today getting ignored and only show events starting tomorrow? Been tryin around but I couldn´t get that done.
Thanks.

cschwarz007, 2017-09-19:
I tried to do some debugging. It seems node_helper returns "CALENDAR_EVENTS" but does not fill the payload with any events. Hence, the error must be somewhere in calendarfetcher.js. How can I debug that file?

cschwarz007, 2017-09-20:
I think I found the issue. I have a calendar for birthdays which used to display fine. However, since recently it does not work anymore.

This is an extract from the iCal data:

`  { type: 'VEVENT',
     params: [],
     start: { 1956-09-26T22:00:00.000Z tz: undefined },
     end: { 1956-09-27T22:00:00.000Z tz: undefined },
     rrule: 
      RRule {
        _string: null,
        _cache: null,
        origOptions: [Object],
        options: [Object],
        timeset: [Object] },
     dtstamp: '20170920T064041Z',
     uid: '146f6fognqb5c9fn0b5sfdio6c@google.com',
     created: '20170430T173625Z',
     description: '',
     'last-modified': '20170430T173625Z',
     location: '',
     sequence: '0',
     status: 'CONFIRMED',
     summary: 'Ramona Schöntaube\'s Birthday',
     transparency: 'TRANSPARENT' },
  'vvhvn1l1l8c5vcarqekr96pvng@google.com': `

As you can see, the start and end date are prior to 1970-01-01. This apparently is what causes an uncaught exception:

`Error: dates lower than Thu Jan 01 1970 00:00:00 GMT+0100 (CET) are not supported
    at Object.toOrdinal (/home/pi/MagicMirror/node_modules/rrule-alt/lib/rrule.js:105:15)
    at Iterinfo.rebuild (/home/pi/MagicMirror/node_modules/rrule-alt/lib/rrule.js:1409:35)
    at RRule._iter (/home/pi/MagicMirror/node_modules/rrule-alt/lib/rrule.js:978:10)
    at RRule.between (/home/pi/MagicMirror/node_modules/rrule-alt/lib/rrule.js:794:21)
    at /home/pi/MagicMirror/modules/default/calendar/calendarfetcher.js:122:24
    at Request._callback (/home/pi/MagicMirror/modules/default/calendar/vendor/ical.js/node-ical.js:11:5)
    at Request.self.callback (/home/pi/MagicMirror/node_modules/request/request.js:186:22)
    at emitTwo (events.js:106:13)
    at Request.emit (events.js:191:7)
    at Request.<anonymous> (/home/pi/MagicMirror/node_modules/request/request.js:1163:10)`

Any ideas on what to do - except changing the original iCal data which is a workaround, but not a nice one?

Any help highly appreciated!

Many thanks,
Christian

cschwarz007, 2017-09-20:
Apologies - just found that issue #839 already exists with this topic. Hence, consider this one closed.

closed 2017-09-20 by cschwarz007

nhubbard, 2017-09-20:
@cschwarz007 Yeah, it's a restriction of UNIX (and by extension, Linux) because the time system on these OS's starts at 00:00:01 on January 1, 1970. Dates before that are considered invalid because time before then is negative, and the computer knows that time cannot be negative.