MagicMirrorOrg/MagicMirror #984: npm start issue: electron missing

dakoller, 2017-08-06:
my npm-debug.log says:

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/nodejs', '/usr/bin/npm', 'start' ]
2 info using npm@3.10.10
3 info using node@v6.11.2
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.1.2~prestart: magicmirror@2.1.2
6 silly lifecycle magicmirror@2.1.2~prestart: no script for prestart, continuing
7 info lifecycle magicmirror@2.1.2~start: magicmirror@2.1.2
8 verbose lifecycle magicmirror@2.1.2~start: unsafe-perm in lifecycle true
9 verbose lifecycle magicmirror@2.1.2~start: PATH: /usr/lib/node_modules/npm/bin/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/home/pi/env/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
10 verbose lifecycle magicmirror@2.1.2~start: CWD: /home/pi/MagicMirror
11 silly lifecycle magicmirror@2.1.2~start: Args: [ '-c', 'sh run-start.sh' ]
12 silly lifecycle magicmirror@2.1.2~start: Returned: code: 1  signal: null
13 info lifecycle magicmirror@2.1.2~start: Failed to exec start script
14 verbose stack Error: magicmirror@2.1.2 start: `sh run-start.sh`
14 verbose stack Exit status 1
14 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/lib/utils/lifecycle.js:255:16)
14 verbose stack     at emitTwo (events.js:106:13)
14 verbose stack     at EventEmitter.emit (events.js:191:7)
14 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/lib/utils/spawn.js:40:14)
14 verbose stack     at emitTwo (events.js:106:13)
14 verbose stack     at ChildProcess.emit (events.js:191:7)
14 verbose stack     at maybeClose (internal/child_process.js:891:16)
14 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
15 verbose pkgid magicmirror@2.1.2
16 verbose cwd /home/pi/MagicMirror
17 error Linux 4.9.35-v7+
18 error argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
19 error node v6.11.2
20 error npm  v3.10.10
21 error code ELIFECYCLE
22 error magicmirror@2.1.2 start: `sh run-start.sh`
22 error Exit status 1
23 error Failed at the magicmirror@2.1.2 start script 'sh run-start.sh'.
23 error Make sure you have the latest version of node.js and npm installed.
23 error If you do, this is most likely a problem with the magicmirror package,
23 error not with npm itself.
23 error Tell the author that this fails on your system:
23 error     sh run-start.sh
23 error You can get information on how to open an issue for this project with:
23 error     npm bugs magicmirror
23 error Or if that isn't available, you can get their info via:
23 error     npm owner ls magicmirror
23 error There is likely additional logging output above.
24 verbose exit [ 1, true ]

Please only submit reproducible issues. 

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)
Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version *and* your hardware (Raspberry Pi 2/3, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 0.12.13 or later.

**MagicMirror Version**: Now that the versions have split, tell us if you are using the PHP version (v1) or the newer JavaScript version (v2).

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

dakoller, 2017-08-06:
pi@magicmirror:~ $ cd MagicMirror/
pi@magicmirror:~/MagicMirror $ npm install eletron

pi@magicmirror:~/MagicMirror $ npm install electron

> electron@1.6.11 postinstall /home/pi/MagicMirror/node_modules/electron
> node install.js

magicmirror@2.1.2 /home/pi/MagicMirror
├── electron@1.6.11 
└── UNMET PEER DEPENDENCY grunt@1.0.1

npm WARN grunt-yamllint@0.2.0 requires a peer of grunt@~0.4.0 but none was installed.
pi@magicmirror:~/MagicMirror $ npm install grunt
[     .............] | fetchMetadata: sill resolveWithNewModule lodash@4.3.0 ch

Obviously grunt as a dependency could not be loaded

maxanier, 2017-08-13:
I get the same error message, however, electron is installed

MichMich, 2017-09-29:
Closed due to inactivity.

closed 2017-09-29 by MichMich