MagicMirrorOrg/MagicMirror #985: Feature: Add noon for compliments

mueller-ma, 2017-08-07:
It would be nice to have seperate greetings for noon, e.g. from 11:30 to 12:30

roramirez, 2017-08-07:
Are you interested in develop that feature?

For me  it sound nice! ... but instead of created a new type of the part of day should be add a option of range hours like this

```js
config: {
	compliments: {
	               range: [
                         ["10:30", "11:00", "My compliment 1"],
                         ["12:30", "13:00", "Dinner time"]
                 ]
	}
}
```

MichMich, 2017-09-29:
Feel free to send a PR. Closed due to inactivity.

closed 2017-09-29 by MichMich

Stromwerk, 2017-10-13:
I would like to try my hand at adding this functionality. Should I just submit a PR when I've got it working and passing the tests or is this considered a big enough change that I should discuss my solution first, as stated in the CONTRIBUTING.md ?

MichMich, 2017-10-13:
I think it’s good to discuss it first since it would be a change which can be done in a lot of different ways. The idea propped by @roramirez is still pretty limited. Maybe a cron-type of time notation would be better. This way you can also set specific messages for a specific day. 

Better yet would be an implementation in which we can use multiple notations. :)

Whatever implementation we choose: it must be a non breaking change. 

Thanks for your contribution. I appreciate it.