MagicMirrorOrg/MagicMirror #988: Feature: Change background from black to image/slideshow

metalshreds, 2017-08-10:
I saw this project (https://dakboard.com/site) which looks to be exactly magic mirror2  the only difference is the interface to customize the modules and the ability to select the background. 

I am very interested in finding/creating a module that would allow the ability change background from black to slideshow(dropbox possibly or local). I am also learning javascript, but know some html/css.  Does this feature exist or any tips on getting started creating it?
Thanks!

ToddKlindt, 2017-08-10:
There are several modules that will show photos from various locations. You can see them here, https://github.com/MichMich/MagicMirror/wiki/MagicMirror%C2%B2-Modules#3rd-party-modules. I personally use the MMM-RandomPhoto module and it works really well.

roramirez, 2017-08-11:
@metalshreds To me sound and will be nice if we can add a feature like this. 

Should be as simplest possible and with a non break change in the core. If you can add and maintenance in the future. 

The last word if we can include some like this... comes from @MichMich

metalshreds, 2017-08-12:
That would be great if it was added to core. I'd be happy to help but as I said before, I have limited knowledge of JS, just starting to pick it up and figured working on a module would be a good starting place :) I'm still trying to see how all the files are working together.

fewieden, 2017-08-12:
I'd prefer to stick with a custom module and not add it to the core

metalshreds, 2017-08-13:
Sounds good, I'll take a crack at creating the module. Thanks!

metalshreds, 2017-08-13:
In case anybody else is looking for this functionality. I found this module which is working really well! 

https://github.com/Ultimatum22/MMM-RandomBackground

closed 2017-08-13 by metalshreds