MagicMirrorOrg/MagicMirror #994: Auto config for client.

KnottyCord, 2017-08-17:
I'm not good at coding, but I know just a bit to understand some stuff. But since I made my first mirror a lot of people ask me if I could make them one. But the problem is that the one command installation is not user frendly. Like if you want to connect your mirror to a WiFi you need to go in the files a change the variable, with a keyboard.
If there is someone out there willing to create a full on command installation where you can name you mirror install Module. Move module. Edit wifi passord. etc.... and add like a gear icon at the bottom to acces the setting.

Thank you
Shawn Mercier

Terrib1e, 2017-09-03:
Sure I'm willing to work with you on that.

nhubbard, 2017-09-13:
@KnottyCard Feel free! You could also extend the install script with a WiFi setup option.

KnottyCord, 2017-09-14:
@Terrib1e Sure when can we speek? I have a very basic knownledge on programing. HTML5 And css are what im leaning crurently.

MichMich, 2017-09-29:
Please continue this discussion on http://forum.magicmirror.builders 
I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2017-09-29 by MichMich