MagicMirrorOrg/MagicMirror #995: Error: listen EADDRINUSE :::8080

lesajo, 2017-08-19:
Hi.. i have a little problem. but it works.

 node serveronly/
Starting MagicMirror: v2.1.2
Loading config ...
Loading module helpers ...
No helper found for module: alert.
Initializing new module helper ...
Module helper loaded: updatenotification
No helper found for module: clock.
Initializing new module helper ...
Module helper loaded: calendar
No helper found for module: compliments.
No helper found for module: MMM-WunderGround24Hours.
Initializing new module helper ...
Module helper loaded: MMM-Fuel
Initializing new module helper ...
Module helper loaded: MMM-Wunderlist
No helper found for module: weatherforecast.
Initializing new module helper ...
Module helper loaded: newsfeed
All module helpers loaded.
Starting server on port 8080 ...
Server started ...
Connecting socket for: updatenotification
Connecting socket for: calendar
Starting node helper for: calendar
Connecting socket for: MMM-Fuel
Starting module helper: MMM-Fuel
Connecting socket for: MMM-Wunderlist
Connecting socket for: newsfeed
Starting module: newsfeed
Sockets connected & modules started ...

Ready to go! Please point your browser to: http://localhost:8080
Whoops! There was an uncaught exception...
{ Error: listen EADDRINUSE :::8080
    at Object._errnoException (util.js:1041:11)
    at _exceptionWithHostPort (util.js:1064:20)
    at Server.setupListenHandle [as _listen2] (net.js:1322:14)
    at listenInCluster (net.js:1370:12)
    at Server.listen (net.js:1466:7)
    at new Server (/home/pi/MagicMirror/js/server.js:26:9)
    at /home/pi/MagicMirror/js/app.js:219:18
    at loadNextModule (/home/pi/MagicMirror/js/app.js:168:5)
    at /home/pi/MagicMirror/js/app.js:163:6
    at Class.loaded (/home/pi/MagicMirror/modules/node_modules/node_helper/index.js:19:3)
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '::',
  port: 8080 }
MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues

fewieden, 2017-08-19:
Most likely there is an instance of magicmirror still running on port 8080, which wasn't exited properly.

lesajo, 2017-08-19:
Okay. Thx u

roramirez, 2017-08-19:
This kind of request as  issue are recurrently. I think we need move this request to the forum, if is not a bug or feature discusión should be move there. 

What do you think? cc @fewieden @MichMich

closed 2017-08-22 by MichMich

fatcatcgo, 2017-10-28:
Ran into this issue when I got a new DHCP assigned IP.  Change the IP in the whitelist for listener.  Where else is the server starting to look for IP.  BTW we b runs but some modules have issue.

Also noted that this only occurs when a node on my net is up with it old IP.

hsukup1, 2018-05-10:
fewieden:
"Most likely there is an instance of magicmirror still running on port 8080, which wasn't exited properly."

I figured that this was happening, but how do I exit one properly? I am having a difficult time researching this answer.

fewieden, 2018-05-10:
@hsukup1 depends on how you are running mm²

with pm2 i think it is `pm2 stop mm`, but i never used pm2 so it might not working.

without pm2 you can check with `lsof` what is using your port and terminate it

`lsof -i:8080` and then `kill -9 PID`