MagicMirrorOrg/MagicMirror #1001: Alexa Display Cards

dylono96, 2017-08-22:
is it possible to get the output of the alexa app displayed on the mirror? 
I saw that some other guy made something, but that was not build for the magic mirror 2.

D3rpi, 2017-09-03:
That's something I have been wanting to do with my set up for a while now but sadly don't know how to do that.
Do you have a link for the thing the guy made?

dylono96, 2017-09-04:
this is the link I first saw it on: https://www.hackster.io/mexitek/magic-mirror-amazon-echo-and-aws-iot-29bba5

dylono96, 2017-09-13:
so i found another thing: https://github.com/joanaz/MirrorMirrorOnTheWallSkill maybe you should give that a try.

nhubbard, 2017-09-13:
@dylono96 You're welcome to build a module with AWS Lambda with an Amazon Echo, but we can't show cards from the Alexa app (it's a private API restricted by a usage agreement).

nhubbard, 2017-09-13:
@MichMich close

dylono96, 2017-09-14:
So even with acces to that api it isnt possible because it has anotger license?

dylono96, 2017-09-19:
ok I understand, but could you explain how the MirrorMirrorOnTheWall skill works then?

nhubbard, 2017-09-19:
@dylono96 That skill works as I said (you can only access apps you create) because as part of the setup, you create a skill on AWS Lambda and attach it to your MM.

MichMich, 2017-09-29:
Please refer to http://forum.magicmirror.builders for support, troubleshooting and module specific discussions.

I'm closing this issue for now. If you feel this issue should remain open, please let me know.

closed 2017-09-29 by MichMich