MagicMirrorOrg/MagicMirror #1002: Blank screen when I use module MMM-Wunderlist

VisualTwo, 2017-08-24:
Hello friends,

I tried several times and receive the same errors every time.

When I start with the correctly configured module MMM-Wunderlist, the Magic Mirror blanks.
Here are the pm2 logs:

WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module 'wunderlist'
App threw an error during load
Error: Cannot find module 'wunderlist'
    at Module._resolveFilename (module.js:470:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._load (module.js:418:25)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-Wunderlist/fetcher.js:10:21)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-Wunderlist/fetcher.js:118:3)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32)
{ Error: Cannot find module 'wunderlist'
    at Module._resolveFilename (module.js:470:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._load (module.js:418:25)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-Wunderlist/fetcher.js:10:21)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-Wunderlist/fetcher.js:118:3)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32) code: 'MODULE_NOT_FOUND' }

My Raspberry is on Debian Stretch.
Any solution?

VisualTwo, 2017-08-24:
Here is the part of the config.js - when I disable it, the screen shows the expected contents:

```
                {
                        module: 'MMM-Wunderlist',
                        position: 'top_left',
                        disabled: false,
                        header: 'Wunderlist',
                        config: {
                                accessToken: ">>>MYACCESSTOKEN<<<",
                                clientID: ">>>MYCLIENTID<<<",
                                lists: ["Familie"]
                        }
                },
```

roramirez, 2017-08-24:
Please, Refer to [Forum](https://forum.magicmirror.builders) for support and troubleshooting.

cc @MichMich close it.

closed 2017-08-24 by MichMich