MagicMirrorOrg/MagicMirror #1003: Calender timeformat

ostfilinchen, 2017-08-25:
Hi!

i found an issue in the Calender timeformat display. The wrong entry is in calender.js Line 84.

It says, if timeformat=24 then "hh:mm" but the right format is "HH:mm".

Can u change the Line with the next update?

Regards
Daniel

fewieden, 2017-08-25:
It is fixed since 1.5 months or so https://github.com/MichMich/MagicMirror/commit/db0bd3fa2ded2d4ed22670c304f3b7b0a0264bd9#diff-b26c2d32b9ae09e625122f9ae285d361

So it will be in the next update :)

Pretowner, 2017-09-13:
Hi, maybe the same issue: the first entry within the calendar shows up with e.g. 7:00, but it's 19:00 (I'm using 24h settings ( see picture)
btw, can I setup the calendar that the first entry is also shown with the date only (not "friday...)?

Thanks!
[image]

MichMich, 2017-09-29:
Fixe din next release. Feel free to reopen if it doesn't solve the issue.

closed 2017-09-29 by MichMich