MagicMirrorOrg/MagicMirror #1005: Electron failed to install

Spikeboybebpop, 2017-08-25:
I tried to start the program after going through all the steps on the guides. When I did I got the follow. I also have the debug as well. I tried to look up solutions for electron error but couldn't find anything.

I also have the latest verison of RASPBIAN STRETCH WITH DESKTOP and I used the auto install of Magic mirror.

 Any help is appreciated 

pi@raspberrypi:~ $ cd $HOME/MagicMirror
pi@raspberrypi:~/MagicMirror $ npm start

> magicmirror@2.1.2 start /home/pi/MagicMirror
> sh run-start.sh

/home/pi/MagicMirror/node_modules/electron/index.js:9
  throw new Error('Electron failed to install correctly, please delete node_modules/electron and try installing again')
  ^

Error: Electron failed to install correctly, please delete node_modules/electron and try installing again
    at Object.<anonymous> (/home/pi/MagicMirror/node_modules/electron/index.js:9:9)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/home/pi/MagicMirror/node_modules/electron/cli.js:3:16)
    at Module._compile (module.js:570:32)

npm ERR! Linux 4.9.41-v7+
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! node v6.11.2
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.1.2 start: `sh run-start.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the magicmirror@2.1.2 start script 'sh run-start.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     sh run-start.sh
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log

Debug
  
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/nodejs', '/usr/bin/npm', 'start' ]
2 info using npm@3.10.10
3 info using node@v6.11.2
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.1.2~prestart: magicmirror@2.1.2
6 silly lifecycle magicmirror@2.1.2~prestart: no script for prestart, continuing
7 info lifecycle magicmirror@2.1.2~start: magicmirror@2.1.2
8 verbose lifecycle magicmirror@2.1.2~start: unsafe-perm in lifecycle true
9 verbose lifecycle magicmirror@2.1.2~start: PATH: /usr/lib/node_modules/npm/bin/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
10 verbose lifecycle magicmirror@2.1.2~start: CWD: /home/pi/MagicMirror
11 silly lifecycle magicmirror@2.1.2~start: Args: [ '-c', 'sh run-start.sh' ]
12 silly lifecycle magicmirror@2.1.2~start: Returned: code: 1  signal: null
13 info lifecycle magicmirror@2.1.2~start: Failed to exec start script
14 verbose stack Error: magicmirror@2.1.2 start: `sh run-start.sh`
14 verbose stack Exit status 1
14 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/lib/utils/lifecycle.js:255:16)
14 verbose stack     at emitTwo (events.js:106:13)
14 verbose stack     at EventEmitter.emit (events.js:191:7)
14 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/lib/utils/spawn.js:40:14)
14 verbose stack     at emitTwo (events.js:106:13)
14 verbose stack     at ChildProcess.emit (events.js:191:7)
14 verbose stack     at maybeClose (internal/child_process.js:891:16)
14 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
15 verbose pkgid magicmirror@2.1.2
16 verbose cwd /home/pi/MagicMirror
17 error Linux 4.9.41-v7+
18 error argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
19 error node v6.11.2
20 error npm  v3.10.10
21 error code ELIFECYCLE
22 error magicmirror@2.1.2 start: `sh run-start.sh`
22 error Exit status 1
23 error Failed at the magicmirror@2.1.2 start script 'sh run-start.sh'.
23 error Make sure you have the latest version of node.js and npm installed.
23 error If you do, this is most likely a problem with the magicmirror package,
23 error not with npm itself.
23 error Tell the author that this fails on your system:
23 error     sh run-start.sh
23 error You can get information on how to open an issue for this project with:
23 error     npm bugs magicmirror
23 error Or if that isn't available, you can get their info via:
23 error     npm owner ls magicmirror
23 error There is likely additional logging output above.
24 verbose exit [ 1, true ]

Little-John21, 2017-08-27:
I was having a similar issue where the dependencies wouldn't install but as soon as i got the warnings i hit " -g " enter in terminal and it downloaded them and all was fine.

nhubbard, 2017-08-28:
What kind of Raspberry Pi are you running on? 1, 2, 3 or Zero?

ruicky, 2017-08-30:
I have the same mistake.
I have network restrictions in china. So, I'm using Taobao registory, like ：
`npm i electron  --registry=https://registry.npm.taobao.org`
I hope I can help you。

rhysxevans, 2017-08-30:
Hi

I had the same issue had to change the node version in the script from 6.x to 7.x to get it to work.

so 
1) wget https://raw.githubusercontent.com/MichMich/MagicMirror/master/installers/raspberry.sh
2) edit raspberry.sh  , search for NODE_STABLE_BRANCH="6.x" replace with NODE_STABLE_BRANCH="7.x"
3) save file
4) run raspberry.sh

I am assuming here you have cleaned up and are running from a fresh install

Thanks

jzyee, 2017-09-24:
i have changed the raspberry.sh but still end up with an electron error.

Error: Failed to find Electron v1.7.8

hwikene, 2017-09-24:
I have the same issue.. can someone please show a solution that fixes this?

MichMich, 2017-09-29:
This is more of an NPM or Electron issue. Please refer to those repositories.

closed 2017-09-29 by MichMich

lakario, 2017-09-29:
@MichMich Do you have any idea what's wrong with the installation of these other components? Myself and several others can't get past the RPI setup script.

lakario, 2017-09-29:
Got it working @MichMich. A thread in the MagicMirror Troubleshooting forum pointed out an incompatibility in the latest version of electron (1.7.8). If your install fails, delete the electron folder from node_modules in MirrorMirror, make sure a newer Electron is not already installled globally, and then run:

`sudo npm install -g electron@1.7.6`

nhubbard, 2017-09-29:
@lakario This appears to be a working solution. Another Electron app I am working on has this same exact issue.

yuen33, 2017-11-21:
I also got exactly this error... Tried the solution from @ruicky that doesn't solve my problem...

JamieH1986, 2018-01-22:
You superstar I was ready for tearing my hair out!