MagicMirrorOrg/MagicMirror #1007: Npm start failing

sharmmoh1983, 2017-08-29:
/MagicMirror $ npm start

> magicmirror@2.1.2 start /home/pi/MagicMirror
> sh run-start.sh

npm ERR! Linux 4.4.50-v7+
npm ERR! argv "/opt/nodejs/bin/node" "/usr/local/bin/npm" "start"
npm ERR! node v7.2.1
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! magicmirror@2.1.2 start: `sh run-start.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the magicmirror@2.1.2 start script 'sh run-start.sh'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the magicmirror package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     sh run-start.sh
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs magicmirror
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls magicmirror
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/MagicMirror/npm-debug.log

ivuorinen, 2017-08-31:
> npm ERR! Please include the following file with any support request:
> npm ERR! /home/pi/MagicMirror/npm-debug.log

Without the contents of that file, debugging and helping you solve the problem will be really hard. 
Could you please include it here?

sharmmoh1983, 2017-08-31:
This is the content of the required file:
0 info it worked if it ends with ok
1 verbose cli [ '/opt/nodejs/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@3.10.10
3 info using node@v7.2.1
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.1.2~prestart: magicmirror@2.1.2
6 silly lifecycle magicmirror@2.1.2~prestart: no script for prestart, continuing
7 info lifecycle magicmirror@2.1.2~start: magicmirror@2.1.2
8 verbose lifecycle magicmirror@2.1.2~start: unsafe-perm in lifecycle true
9 verbose lifecycle magicmirror@2.1.2~start: PATH: /opt/nodejs/lib/node_modules/npm/bin/node-gyp-bin:/home/pi/MagicMirror/node_modules/.bin:/opt/nodejs/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/b$
10 verbose lifecycle magicmirror@2.1.2~start: CWD: /home/pi/MagicMirror
11 silly lifecycle magicmirror@2.1.2~start: Args: [ '-c', 'sh run-start.sh' ]
12 silly lifecycle magicmirror@2.1.2~start: Returned: code: 1  signal: null
13 info lifecycle magicmirror@2.1.2~start: Failed to exec start script
14 verbose stack Error: magicmirror@2.1.2 start: `sh run-start.sh`
14 verbose stack Exit status 1
14 verbose stack     at EventEmitter.<anonymous> (/opt/nodejs/lib/node_modules/npm/lib/utils/lifecycle.js:255:16)
14 verbose stack     at emitTwo (events.js:106:13)
14 verbose stack     at EventEmitter.emit (events.js:191:7)
14 verbose stack     at ChildProcess.<anonymous> (/opt/nodejs/lib/node_modules/npm/lib/utils/spawn.js:40:14)
14 verbose stack     at emitTwo (events.js:106:13)
14 verbose stack     at ChildProcess.emit (events.js:191:7)
14 verbose stack     at maybeClose (internal/child_process.js:885:16)
14 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
15 verbose pkgid magicmirror@2.1.2
16 verbose cwd /home/pi/MagicMirror
17 error Linux 4.4.50-v7+
18 error argv "/opt/nodejs/bin/node" "/usr/local/bin/npm" "start"
19 error node v7.2.1
20 error npm  v3.10.10
21 error code ELIFECYCLE
22 error magicmirror@2.1.2 start: `sh run-start.sh`
22 error Exit status 1
23 error Failed at the magicmirror@2.1.2 start script 'sh run-start.sh'.
23 error Make sure you have the latest version of node.js and npm installed.
23 error If you do, this is most likely a problem with the magicmirror package,
23 error not with npm itself.
23 error Tell the author that this fails on your system:
23 error     sh run-start.sh
23 error You can get information on how to open an issue for this project with:
23 error     npm bugs magicmirror
23 error Or if that isn't available, you can get their info via:
23 error     npm owner ls magicmirror
23 error There is likely additional logging output above.
24 verbose exit [ 1, true ]

sharmmoh1983, 2017-08-31:
I have been trying very hard to crack this ...but not able to understand the issue

Raspy94, 2017-09-02:
If someone finds a solution could please post it here?

D3rpi, 2017-09-02:
I have the same issue. This is the contents of my debug file. I'm still on Jessie. 

> 
> 0 info it worked if it ends with ok
> 1 verbose cli [ '/usr/bin/nodejs', '/usr/bin/npm', 'start' ]
> 2 info using npm@1.4.21
> 3 info using node@v0.10.29
> 4 verbose run-script [ 'prestart', 'start', 'poststart' ]
> 5 info prestart magicmirror@2.1.2
> 6 info start magicmirror@2.1.2
> 7 verbose unsafe-perm in lifecycle true
> 8 info magicmirror@2.1.2 Failed to exec start script
> 9 error magicmirror@2.1.2 start: `sh run-start.sh`
> 9 error Exit status 127
> 10 error Failed at the magicmirror@2.1.2 start script.
> 10 error This is most likely a problem with the magicmirror package,
> 10 error not with npm itself.
> 10 error Tell the author that this fails on your system:
> 10 error     sh run-start.sh
> 10 error You can get their info via:
> 10 error     npm owner ls magicmirror
> 10 error There is likely additional logging output above.
> 11 error System Linux 4.9.35-v7+
> 12 error command "/usr/bin/nodejs" "/usr/bin/npm" "start"
> 13 error cwd /home/drpi1/MagicMirror
> 14 error node -v v0.10.29
> 15 error npm -v 1.4.21
> 16 error code ELIFECYCLE
> 17 verbose exit [ 1, true ]
>

roramirez, 2017-09-03:
Did you run `npm install`?

fewieden, 2017-09-03:
@D3rpi you should definetly update your node version first, as yours is over 3 years old `3 info using node@v0.10.29`

Raspy94, 2017-09-03:
I have a similar problem but my npm and node are updated 
`root@raspberrypi:/home/pi/MagicMirror# DISPLAY=:0 npm start

> magicmirror@2.1.2 start /home/pi/MagicMirror
> sh run-start.sh

No protocol specified
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! magicmirror@2.1.2 start: `sh run-start.sh`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the magicmirror@2.1.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2017-09-03T08_05_14_282Z-debug.log`

`
0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@5.4.0
3 info using node@v8.4.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle magicmirror@2.1.2~prestart: magicmirror@2.1.2
6 info lifecycle magicmirror@2.1.2~start: magicmirror@2.1.2
7 verbose lifecycle magicmirror@2.1.2~start: unsafe-perm in lifecycle true
8 verbose lifecycle magicmirror@2.1.2~start: PATH: /usr/local/lib/node_modules/npm/bin/node-gyp-bin:/home/pi/MagicMirror/n$
9 verbose lifecycle magicmirror@2.1.2~start: CWD: /home/pi/MagicMirror
10 silly lifecycle magicmirror@2.1.2~start: Args: [ '-c', 'sh run-start.sh' ]
11 silly lifecycle magicmirror@2.1.2~start: Returned: code: 1  signal: null
12 info lifecycle magicmirror@2.1.2~start: Failed to exec start script
13 verbose stack Error: magicmirror@2.1.2 start: `sh run-start.sh`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:280:$
13 verbose stack     at emitTwo (events.js:125:13)
13 verbose stack     at EventEmitter.emit (events.js:213:7)
13 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:$
13 verbose stack     at emitTwo (events.js:125:13)
13 verbose stack     at ChildProcess.emit (events.js:213:7)
13 verbose stack     at maybeClose (internal/child_process.js:927:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
14 verbose pkgid magicmirror@2.1.2
15 verbose cwd /home/pi/MagicMirror
16 verbose Linux 4.9.35-v7+
17 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
18 verbose node v8.4.0
19 verbose npm  v5.4.0
20 error code ELIFECYCLE
21 error errno 1
22 error magicmirror@2.1.2 start: `sh run-start.sh`
22 error Exit status 1
23 error Failed at the magicmirror@2.1.2 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]`

D3rpi, 2017-09-03:
@fewieden Thanks.
I updated my node and it's working again. I'm not sure what caused the issue because it was working fine until last week when I updated my Pi3(still on jessie).

roramirez, 2017-09-04:
@fewieden Maybe we need to check the version of Node before to start the electron app, What do you think?

sharmmoh1983, 2017-09-05:
Even after updating node version to 7 still I ma getting this issue

sharmmoh1983, 2017-09-19:
It is working now

closed 2017-09-19 by sharmmoh1983