MagicMirrorOrg/MagicMirror #1009: Feature Request: Module Schedules

rhysxevans, 2017-08-30:
Hi

Firstly thanks for the project.

I was wondering if it would be possible to a schedule to the module config sections.

What I mean is the following 

From 05:00 - 10:00 I would like module X to run at position A
From 10:00 - 15:00 I would like module Y to run at position A

This would allow the screen to "dynamically" update with relevant info based on the time of day (could possibly go further and also do it by day)

I am thinking along the lines of between 01:00 - 05:00 I don't need to know the bus times (as they don't run) but I might want to know the temperature. Or in the morning I need to know about trains leaving from X station but in the afternoon I need to know about trains leaving from Y station.

I hope that makes sense ?

I am not a programmer so wouldn't know where to start with this. I could do a crude cron job that swaps out he config files but that feels "wrong"

Thoughts

Thanks

rhysxevans, 2017-08-30:
Aargh. just found this https://github.com/ianperrin/MMM-ModuleScheduler

Why couldn't I find it before !

Apologies for wasting your time

roramirez, 2017-08-31:
Please close the issue

cc @MichMich

closed 2017-08-31 by rhysxevans