MagicMirrorOrg/MagicMirror #1017: Weather forecast module doesn't show up

rhodso, 2017-09-07:
**Platform**: Raspberry Pi Model 3b
**Node Version**: Latest
**MagicMirror Version**: Javascript

**Description**: 
As discused in: https://forum.magicmirror.builders/topic/4781/problem-with-weather-forecast/16?page=1 , the Weather forecast module doesn't show up. It has never showed up ever since my install 2 days ago

**Steps to Reproduce**: Set up on a Pi using the automatic installer
**Expected Results**: Weather forecast section
**Actual Results**: Weather forecast section not there
**Configuration**: https://pastebin.com/cf3eHaHM
**Additional Notes**: Attempted the 3h fix from Mykle1, still didn't work

sfmailand, 2017-09-07:
It looks to me like you haven´t put a valid API key into the config, which would cause it to be unable to pull weather (and therefore not be able to display it)

rhodso, 2017-09-08:
Nope, that's not it.
I've tried it with both keys I own and I generated another key.

Config file: https://pastebin.com/xdAwUHnD
WeatherForecast.js file: https://pastebin.com/H6qkhbVg

I verified that all 3 keys I own work by using the Current Weather Module, which shows up evey time I have a valid key

Negendahl, 2017-09-08:
Im having this issue too, i located the problem not being with the API, but the /daily part of the URL. 
This is my "original" URL created by weather forecast (dosent work - Invalid API key error)
http://api.openweathermap.org/data/2.5/forecast/daily?id=2613766&units=metric&lang=en&cnt=7&APPID=APIKEY

This URL works (only removed the "/daily" part of the URL):
http://api.openweathermap.org/data/2.5/forecast?id=2613766&units=metric&lang=en&cnt=7&APPID=APIKEY

I cant get to to work though the module though, but the problem is not with the API key.

Hope it helps :) 

/Mick

nhubbard, 2017-09-08:
@MichMich Could you take a look at this? I'm currently inundated with school, so I can't do it.

MichMich, 2017-09-08:
I just did a clean install of the MagicMirror master branch and used @rhodso's config file using my own API key. The weather forecast appears as expected.

Just to be sure:

- Which branch are you running?
- Are you running the latest version? (`git pull` to update).
- Did you try removing the node_modules folder and run `npm install` again?
- Do you see any errors in the Electron console?
- Are you sure used the correct API key?

Let me know!

acejacek, 2017-09-09:
Same problem with me. Latest pull from master branch, run in "server only" mode. @MichMich, create a new API key, maybe `forecast/daily` is not available for users with newer key? Or better, test mine key, I'll keep it valid for some time, specially for tests: d3ac9535fb87ec656c8d1ece569b877c

djzwieback, 2017-09-10:
Check out the openweather API description:
https://openweathermap.org/api
For the daily forecast it requires a paid account.

MichMich, 2017-09-10:
@acejacek @djzwieback I think you guys are right. I tried it with both acejacek's API key as well as a newly created one (new account), and both get an unauthorised. I think we need to start looking for an alternative API source.

nhubbard, 2017-09-10:
@MichMich I would suggest using Dark Sky. My account on OpenWeatherMap still works (for now) but might stop randomly working in the future.

nhubbard, 2017-09-10:
@djzwieback Great. They didn't even notify us... I have my email on there, and I didn't even get a notification. :-1:

robmcc1983, 2017-09-10:
My weather forecast isn't working either and as others have said, Api key works with current weather in config but not with forecast.
I just get no weather at all appear when using weather forecast..
Cheers

robmcc1983, 2017-09-10:
According to openweather 5 day forecast and Ali updates every 2 hours is still available for free users.
As default forecast is set to 7 days.
I'm no expert but could the forecast work if we set our config to 5 day forecast and update interval to 2 hours?

lgreaves, 2017-09-12:
The API endpoint: "api.openweathermap.org/data/2.5/forecast/daily" is restricted for new accounts.  I get code 401 "Invalid API key" for the account I just created.  I've raised a support request with openweathermaps, hopefully it's a provisioning error!!

If your account was created sometime ago (my account was created last year) the endpoint "/forecast/daily" works under the free plan.  

It should be noted that the endpoint "api.openweathermap.org/data/2.5/forecast" works for both of my accounts (new and old).  Can the weather module be updated to use the API endpoint "/forecast" instead of "/forecast/daily"?

MichMich, 2017-09-12:
@lgreaves Feel free to send a PR to the develop branche.

rhodso, 2017-09-13:
I've tried replacing the line where the url is built with:

`//var url = this.config.apiBase + this.config.apiVersion + "/" + this.config.forecastEndpoint + `this.getParams();`
`var url = "http://api.openweathermap.org/data/2.5/forecast/daily?q=Stoke-on-Trent,GB&units=" + this.config.units + "&lang=" + this.config.lang + "&APPID=[MYAPI]"`

Does that look right?

rhodso, 2017-09-13:
I've just tried the following:
http://api.openweathermap.org/data/2.5/forecast?q=Stoke-on-Trent,GB&appid=[My API]

And that gives me the weather forecast, but when I try replacing the url with that it still doesn't show up. 

I've tried using the default config, then modifying the line as listed in my previous comment, then changing the endpoint to forecast instead of forecast/data

Is there like a quick fix I can use that will just make it work until the issue is fixed?

nhubbard, 2017-09-13:
Okay, @MichMich. I figured it out. New accounts on OpenWeatherMap don't get access to the daily endpoint. It is now paid, *but* us original users were silently grandfathered into the ability to use it for free.

For now, it seems like a good idea to start thinking about re-writing the weather widget in Dark Sky. Dark Sky won't make it paid, because they have revenue from outside of the API (their mobile apps are paid).

luizcarlosnery, 2017-09-14:
Same personal problem.
Mm2 does not display the Weatherforecast on the screen.

kmcrawford, 2017-09-19:
The Hack fix worked for me. 

https://github.com/roramirez/MagicMirror/commit/58b763e935a469d9912c1953c0a3d0af8d2d67b5

preetv, 2017-09-24:
My weather forecast module doesn't show anything but it only shows up i.e loading. What to do . And why is it not working. Your reply will matter me the most.

Sp4M, 2017-09-24:
Hi, the problem was discussed and partially solved in the forum
https://forum.magicmirror.builders/topic/4781/problem-with-weather-forecast/33

preetv, 2017-09-25:
Ok now if i want to change the clock's time and i want to set the time of my country how is it possible. Also how can i add or run google voice control module?

MichMich, 2017-09-29:
Fixed: https://github.com/MichMich/MagicMirror/commit/29f3a81666829cab08f24bc49336befc4ae2df15

closed 2017-09-29 by MichMich