MagicMirrorOrg/MagicMirror #1018: Black Screen Issue

Nograe, 2017-09-10:
Hello,
After the setup of config.js and installation of npm and node, I succeed to launch the app but the screen is all black, and same for the index.html

Nograe, 2017-09-11:
Just type npm start in vendor file and voilà !

closed 2017-09-11 by Nograe

axellejamous, 2017-10-19:
hello @Nograe I'm having the same issue. Tried all of the other suggestions mentioned in other issues that are similar. including the "en" change but nothing seems to be working so my problem must be elsewhere.

What do you mean by vendor file?

MichMich, 2017-10-19:
@axellejamous Please check http://forum.magicmirror.builders for support.

hyd00, 2017-11-12:
for anyone else, i solved my blank screen issue by

```
cd ~/MagicMirror/vendor
npm install
```
it will download fonts and icons needed by modules.

then
```
cd ~/MagicMirror
npm start
```
and now it shows modules perfectly 💃  ..

bucbowie, 2018-12-15:
+1 and kudos to hyd00

bucbowie, 2018-12-15:
Magic Mirror would not properly install for me after doing apt-get update && apt-get upgrade. However, after doing: apt-get -y dist-upgrade, the Magic Mirror package installed. From there, I went inside MagicMirror and ran: git pull && npm install. That brought me current with an installed version of Magic Mirror. Again, thanks to hyd00 for the vendor folder recommendation.

reillychase, 2019-11-17:
I was stuck for hours on this, thanks @hyd00! Should be added to the instructions for manual install.