MagicMirrorOrg/MagicMirror #1020: Pass module instance to node_helper

sharmmoh1983, 2017-09-13:
Hi 
I am trying to pass module instance to node_helper using send notification methods but no notification get triggered

If I pass this.config everything works..Please let me know how to use require in module rather than node_helper

MichMich, 2017-09-13:
I don't really understand what you mean. Since this isn't a core bug report, I suggest you use the forum for developer support. http://magicmirror.builders.

sharmmoh1983, 2017-09-13:
I have tried forum also but there is no response from them .I  am trying following thing:

I am trying to pass module instance to node helper but in that scenario, no communication happens and node helper received notifications never gets triggered
Module:
this.sendSocketNotification(“CONFIG”, this);
Node helper:
socketNotificationReceived: function(notification, moduleInstance) {
if (notification === “CONFIG”) {
this.activateSonus(moduleInstance)
console.log(“Working notification system. Notification:”, notification, "payload: ", payload);
}
},

sharmmoh1983, 2017-09-13:
Nothing gets triggered while I send the instance. Also I needed some help in using require('sonus') kind of functionality in module . How to achieve it inmodule

MichMich, 2017-09-13:
You can't send a module instance to the node_helper. You can only send JSON objects,

sharmmoh1983, 2017-09-13:
Then how can make use of require packages in module

MichMich, 2017-09-13:
What do you mean with that? You can just use require in the node helper.

sharmmoh1983, 2017-09-13:
Please help me in getting this through as no one is responding on forum for a long time

sharmmoh1983, 2017-09-13:
No I want to use it in the module and not in node_helper. Can I make use of get scripts to get require functionality in modules

MichMich, 2017-09-13:
Yes, thats what the scripts are for. My best advice is to explain it a bit better on the forum. It's unclear what you want to do. Without a good explanation, no-one can help you. Please continue on the forum, since the GitHub issues are specifically for core-bugruports. Thanks!

closed 2017-09-13 by MichMich