MagicMirrorOrg/MagicMirror #1021: Cannot create complex Json structure in module and Logging not working

sharmmoh1983, 2017-09-13:
Hey

I am trying to create complex json object in module and  pass it one of the script included in get script

const commands = {
  'hello': function () {
    console.log('You will obey');
  },
  '(give me) :flavor ice cream': function (flavor) {
    console.log('Fetching some ' + flavor + ' ice ceam for you sr')
  },
  'turn (the)(lights) :state (the)(lights)': function (state) {
    console.log('Turning the lights', (state == 'on') ? state : 'off')
  }
}

annyang.addCommands(commands);

Commands are always turning out to be blank

Also I am trying to use Log statements but nothing comes in console:

Log.log(payload);

sharmmoh1983, 2017-09-13:
Callbacks in json are getting blank  or not get created

MichMich, 2017-09-13:
Hi @sharmmoh1983, as explained in your previous issue, github issues are here for core bug reports. If you need support building your module, please use the forum. We cannot give support via GitHub issues. 

What I can say is what you're trying to do is not possible. Json objects can't contain methods. Please checkout some info on JSON. Google is your friend. 

I urge you to not open any other topics regarding development support. Thanks.

closed 2017-09-13 by MichMich

sharmmoh1983, 2017-09-13:
Ok .. Thank buddy

sharmmoh1983, 2017-09-14:
Just to update you, above thing is working in node_helper so there is no issue in JSON structure as JSON is just map of keys and values.

But some how in module class, it is getting converted it into some other  object other than JSON which is getting failed in method call

That is why I am updating it as it seems to be bug in your module class

MichMich, 2017-09-14:
In the node module it's a JavaScript object. Not a JSON object. JSON objects can only contain strings, numbers, arrays and objects. Not functions.

sharmmoh1983, 2017-09-14:
It can take functions also as happening in node_helper, anyways as per your suggestion I have posted detailed requirement on forum

MichMich, 2017-09-14:
In the node helper it's a JavaScript object. Not a JSON object ...

fewieden, 2017-09-17:
Could you give a reasonable usecase, why someone would need the module instance in the node helper?

sharmmoh1983, 2017-09-17:
Hi

I am trying to communicate with other modules through node helper rather
than sending notification back to the curent module and then sending
notifications to other modules