MagicMirrorOrg/MagicMirror #1027: UpdateDOM throwing Error

sharmmoh1983, 2017-09-19:
Hi
I am trying to show module content on voice command so during start up I am calling
this.hide(0) and in getDOM method checking this.hidden parameter so getDOM does not return anything during startup

start: function() {
Log.info("Starting module: " + this.name);

    requiresVersion: "2.1.0",

    // Set locale.
    this.url = this.getEventsUrl();
    this.event = {};
    this.activeItem = 0;
    this.rotateInterval = null;
    this.hide(0);
   // this.scheduleUpdate();
},

getDom: function() {
Log.log("In getDom events", this.hidden);
   if (!this.hidden) {
    var wrapper = document.createElement("div");
    wrapper.className = "wrapper";
    wrapper.style.maxWidth = this.config.maxWidth;

    if (!this.loaded) {......

socketNotificationReceived: function(notification, payload) {

        if (notification === "EVENTS_RESULT") {
Log.log("socketNotificationReceived EVENTS_RESULT");
            this.processEvents(payload);
            if (this.rotateInterval == null) {
                this.scheduleCarousel();
            }
Log.log("socketNotificationReceived EVENTS_RESULT1");
            this.show(0);
            this.updateDom();
        }
        //this.updateDom();
    },

But I  am getting error while calling this.updateDom();

moduleWrapper in main.js is always coming as null

In getDom events false
main.js:134 module_6_MMM-Events
main.js:136 null
main.js:137 Uncaught TypeError: Cannot read property 'getElementsByClassName' of null
    at moduleNeedsUpdate (main.js:137)
    at updateDom (main.js:104)
    at Object.updateDom (main.js:471)
    at Class.updateDom (module.js:295)
    at Class.socketNotificationReceived (MMM-Events.js:203)
    at module.js:183
    at r.<anonymous> (socketclient.js:25)
    at r.emit (index.js:133)
    at r.onevent (socket.js:270)
    at r.MMSocket.self.socket.onevent (socketclient.js:19)

sharmmoh1983, 2017-09-19:
This is my config file:
modules: [
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		
		
		{
			module: "compliments",
			position: "top_third"
		},
		
		
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "New York Times",
						url: "http://www.nytimes.com/services/xml/rss/nyt/HomePage.xml"
					}
				],
				showSourceTitle: true,
				showPublishDate: true
			}
		},
{
            module: 'MMM-Voice-Control-Sonus',
            config: {
              projectId: 'aimevoice2017',
              keyFilename: '/resources/AimeVoice-91efae41b5b4-1.json',
              hotWord : 'aime',
              hotWordFile : '/resources/aime.pmdl'

            }
        },
{
   module: 'MMM-googlemaps',
   position: 'top_center',
   config: {
       apikey: 'AIzaSg5aknOUo5shOQ',
       origin: '-31.952326,115.857847',
       destination: '-32.00469,115.89405',
       width: "600px", // Optional. Default: 100%
       height: "300px" //Optional. Default: 100px
   }
},

	
/*{
		module: "MMM-EmbedYoutube", // Path to youtube module from modules folder Exmaple: MagicMirror/modules/custom/youtube/ so it's custom/youtube
		position: "bottom_left",	// This can be any of the regions.
		config: {
			// See 'Configuration options' for more information.
			watch: "w3jLJU7DT5E"
		}
	},*/
{
    module: 'MMM-Events',
    position: 'top_left',
    config: {
	city: "Bangalore",	           // Your City
	eventType: "sports",		   // Choose from Events List below
	when: "This Week",                 // "All", "Future", "Past", "Today", "Last Week", "This Week", "Next week", and months by name, e.g. "October"
	mode: "noFrame",                   // "Frame" or "noFrame" (around picture)
	apikey: "8V8gdwJ8WcHQnLJB",
	rotateInterval: 5 * 60 * 1000,     // New Event Appears every 5 minutes
	useHeader: false,	           // Set to true if you want a header
	header: "",
	animationSpeed: 3000,              // Event fades in and out
	picture: true,                     // true, false = no image
    }
},

When I comment out MMM-googlemaps in above config then there is no issue . IS it related to position values??

sharmmoh1983, 2017-09-19:
There is seems to be some issue with position region getting common like top_left and top centre
which results in this kind of issue

MichMich, 2017-09-19:
@sharmmoh1983 Please, please, please use the forum for these kind of issues. This is a not a confirmed bug, nor did you fill in any of the required Issue details. 

Since this is isn't the first time I'm asking you this, please be informed that the next issue which doesn't conform to the issue guidelines will be closed without any response. Thanks!

closed 2017-09-19 by MichMich

sharmmoh1983, 2017-09-19:
Sorry for the inconvenience but How to know whether it qualifies as bug or not as according to my understanding positions in config if are clashing they are causing updateDOM to throw error which is a potential bug