MagicMirrorOrg/MagicMirror #1036: ipwhitelist not working

t4rif4ri, 2017-09-29:
**Platform**: RPI 3 with newest MM build, installed and updated yesterday.

**Description**:  Until yesterday my mirror was working with an older build version. The mirror was accessable trough the internet via my configured ddns service. 
After the update im getting the error, that the ip was blocked. 
I tested with ipwhitelist [] but it isnt working. Additionally i entered 0.0.0.0 to adress wo get the ipwhitelist: [] working, but it didnt fix the problem.

How can i make the mirror accessable from every ip adress?

MichMich, 2017-09-29:
Are you running the develop branch or the master branch?

t4rif4ri, 2017-09-29:
actual Branch is master

MichMich, 2017-09-29:
Could you give the develop branch a try?

    git checkout develop
    git pull
    npm install

t4rif4ri, 2017-09-29:
Ok, this problem is solved, no blocking anymore in shell console, but cant connect anyway. Browser output: (111) Connection refused

MichMich, 2017-09-29:
So, is it solved? Or isn't it? :D If it isn't solved, what message do you see in your node output when trying to connect?

t4rif4ri, 2017-09-29:
Output is not showing any blocked ips, but the connection got refused when i open the link in the browser. Ports are open, like when it was working.

roramirez, 2017-09-29:
There was  fix into the `develop` branch about this issue. Please, check it out
https://github.com/MichMich/MagicMirror/commit/45d597ac49d52e1728b9d40d95b0a7acf2945cbd

So, this issue is really fixed.

As @MichMich said

```
cd MagicMirror
git checkout develop
git pull
rm -frv node_modules
npm install
```

Make sure you can access from the browser and where is running MagicMirror. Are differents IPs?, I mean MM and Browser

t4rif4ri, 2017-09-30:
mirror is accessable from intern network. Also within the mirror itself. From outside i get the connection refused error shown in my browser. Tried different port settings, but cant connect via dyndns from outside. Other ports like webserver on port 80 is still working and accessable from outside.

roramirez, 2017-09-30:
... I presumed this is configuration issue or other not related with MagicMirror itself. Did you set [`address` to `'0.0.0.0'`  in your config.js](https://github.com/MichMich/MagicMirror#configuration)?

What show you the command `netstat -putan` ?

nhubbard, 2017-09-30:
@t4rif4ri We don't (officially) support DynDNS. I would recommend taking your case to the forums.

@MichMich close?

t4rif4ri, 2017-10-01:
Thank you all for your help! After i set 0.0.0.0 to adress its working again. :))

Wird, 2017-10-01:
Same her, the site was live after adding "address": "0.0.0.0", to the config

closed 2017-10-03 by MichMich